JvanKatwijk/dab-cmdline #1: Compile error

Razmuzh, 2017-02-21:
Getting error when trying to compile on RaspberryPi:
Scanning dependencies of target dab_lib
[  2%] Building CXX object CMakeFiles/dab_lib.dir/src/dab-api.cpp.o
In file included from /usr/include/c++/4.9/atomic:38:0,
                 from /root/dab-cmdline/dab-library/./includes/dab-class.h:36,
                 from /root/dab-cmdline/dab-library/src/dab-api.cpp:30:
/usr/include/c++/4.9/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support for the \
  ^
In file included from /usr/include/c++/4.9/atomic:41:0,
                 from /root/dab-cmdline/dab-library/./includes/dab-class.h:36,
                 from /root/dab-cmdline/dab-library/src/dab-api.cpp:30:
/usr/include/c++/4.9/bits/atomic_base.h:74:3: error: ‘constexpr’ does not name a type
   constexpr memory_order

....
etc

JvanKatwijk, 2017-02-21:
This file requires compiler and library support for the ISO C++ 2011
standard.

Razmuzh, 2017-02-22:
Thanks.
Installed 6.2 and its compiling now.

closed 2017-02-22 by Razmuzh

mickaelmonsieur, 2019-01-08:
Hello,
I have the same error.
My version of gcc is 6.3

```
root@player:~/dab-cmdline/example-1/build# cmake .. -DRTLSDR=ON
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.5.3
-- Found librtlsdr: /usr/include, /usr/lib/arm-linux-gnueabihf/librtlsdr.so
-- Found LibSampleRate: 
-- Checking for module 'portaudio-2.0'
--   Found portaudio-2.0, version 19
-- Performing Test PORTAUDIO2_FOUND
-- Performing Test PORTAUDIO2_FOUND - Success
-- Found PORTAUDIO: /usr/include  
-- Configuring done
-- Generating done
-- Build files have been written to: /root/dab-cmdline/example-1/build
root@player:~/dab-cmdline/example-1/build# make
Scanning dependencies of target dab-rtlsdr
[ 12%] Building CXX object CMakeFiles/dab-rtlsdr.dir/root/dab-cmdline/devices/rtlsdr-handler/rtlsdr-handler.cpp.o
[ 25%] Building CXX object CMakeFiles/dab-rtlsdr.dir/main.cpp.o
/root/dab-cmdline/example-1/main.cpp: In function ‘void programnameHandler(std::__cxx11::string, int, void*)’:
/root/dab-cmdline/example-1/main.cpp:100:9: warning: unused variable ‘i’ [-Wunused-variable]
 int16_t i;
         ^
/root/dab-cmdline/example-1/main.cpp: In function ‘int main(int, char**)’:
/root/dab-cmdline/example-1/main.cpp:318:28: error: invalid conversion from ‘void (*)(bool, void*)’ to ‘ensemblename_t {aka void (*)(std::__cxx11::basic_string<char>, int, void*)}’ [-fpermissive]
                            );
                            ^
/root/dab-cmdline/example-1/main.cpp:318:28: error: invalid conversion from ‘void (*)(bool, int16_t, int32_t, void*) {aka void (*)(bool, short int, int, void*)}’ to ‘programname_t {aka void (*)(std::__cxx11::basic_string<char>, int, void*)}’ [-fpermissive]
/root/dab-cmdline/example-1/main.cpp:318:28: error: invalid conversion from ‘void (*)(std::__cxx11::string, int, void*) {aka void (*)(std::__cxx11::basic_string<char>, int, void*)}’ to ‘fib_quality_t {aka void (*)(short int, void*)}’ [-fpermissive]
/root/dab-cmdline/example-1/main.cpp:318:28: error: invalid conversion from ‘void (*)(std::__cxx11::string, int, void*) {aka void (*)(std::__cxx11::basic_string<char>, int, void*)}’ to ‘audioOut_t {aka void (*)(short int*, int, int, bool, void*)}’ [-fpermissive]
/root/dab-cmdline/example-1/main.cpp:318:28: error: invalid conversion from ‘void (*)(int16_t, void*) {aka void (*)(short int, void*)}’ to ‘dataOut_t {aka void (*)(std::__cxx11::basic_string<char>, void*)}’ [-fpermissive]
/root/dab-cmdline/example-1/main.cpp:318:28: error: invalid conversion from ‘void (*)(int16_t*, int, int, bool, void*) {aka void (*)(short int*, int, int, bool, void*)}’ to ‘bytesOut_t {aka void (*)(unsigned char*, short int, unsigned char, void*)}’ [-fpermissive]
/root/dab-cmdline/example-1/main.cpp:318:28: error: invalid conversion from ‘void (*)(std::__cxx11::string, void*) {aka void (*)(std::__cxx11::basic_string<char>, void*)}’ to ‘programdata_t {aka void (*)(audiodata*, void*)}’ [-fpermissive]
/root/dab-cmdline/example-1/main.cpp:318:28: error: invalid conversion from ‘void (*)(uint8_t*, int16_t, uint8_t, void*) {aka void (*)(unsigned char*, short int, unsigned char, void*)}’ to ‘programQuality_t {aka void (*)(short int, short int, short int, void*)}’ [-fpermissive]
/root/dab-cmdline/example-1/main.cpp:318:28: error: invalid conversion from ‘void (*)(audiodata*, void*)’ to ‘motdata_t {aka void (*)(std::__cxx11::basic_string<char>, int, void*)}’ [-fpermissive]
/root/dab-cmdline/example-1/main.cpp:318:28: error: cannot convert ‘void (*)(int16_t, int16_t, int16_t, void*) {aka void (*)(short int, short int, short int, void*)}’ to RingBuffer<std::complex<float> >*’ for argument ‘14’ to ‘void* dabInit(deviceHandler*, uint8_t, syncsignal_t, systemdata_t, ensemblename_t, programname_t, fib_quality_t, audioOut_t, dataOut_t, bytesOut_t, programdata_t, programQuality_t, motdata_t, RingBuffer<std::complex<float> >*, RingBuffer<std::complex<float> >*, void*)’
/root/dab-cmdline/example-1/main.cpp:368:58: error: invalid conversion from ‘int32_t {aka int}’ to ‘void*’ [-fpermissive]
     programName = dab_getserviceName (serviceId, theRadio);
                                                          ^
In file included from /root/dab-cmdline/example-1/main.cpp:33:0:
/root/dab-cmdline/example-1/../dab-api.h:223:13: note:   initializing argument 1 of ‘std::__cxx11::string dab_getserviceName(void*, int32_t)’
 std::string dab_getserviceName (void *, int32_t);
             ^~~~~~~~~~~~~~~~~~
/root/dab-cmdline/example-1/main.cpp:368:58: error: invalid conversion from ‘void*’ to ‘int32_t {aka int}’ [-fpermissive]
     programName = dab_getserviceName (serviceId, theRadio);
                                                          ^
In file included from /root/dab-cmdline/example-1/main.cpp:33:0:
[... cut ...]
 std::string dab_getserviceName (void *, int32_t);
             ^~~~~~~~~~~~~~~~~~
CMakeFiles/dab-rtlsdr.dir/build.make:86: recipe for target 'CMakeFiles/dab-rtlsdr.dir/main.cpp.o' failed
make[2]: *** [CMakeFiles/dab-rtlsdr.dir/main.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/dab-rtlsdr.dir/all' failed
make[1]: *** [CMakeFiles/dab-rtlsdr.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
root@player:~/dab-cmdline/example-1/build# 

```