JvanKatwijk/dab-cmdline #3: Problems with choosing channel

Razmuzh, 2017-02-23:
In Norway we have some channels with similar name e.g.:
```
NRK P1
NRK P1+
```

If i choose -P "NRK P1" I sometimes get locked to "NRK P1+" if the channel is listet before "NRK P1" in the ensemble.

I also have problems with the Norwegian characters ÆØÅ 

```
program (3): NRK P1+?stfold   in the list  (Should be NRK P1+Østfold)
program (4): NRK P1+?stlands  in the list (Should be NRK P1+Østlandssendingen)
program (5): NRK V?r S?R      in the list (Should be (NRK Vær SØR)
```

But otherwise its working like a sharm!

Thanks a lot!!

R.

JvanKatwijk, 2017-02-23:
I know about the characters, that was ok in the Qt version, but still to be
"solved" in the qt-free version
best
j

Razmuzh, 2017-02-23:
But what about the "autocomplete" - Thats more of a problem.
Is it possible to choose by the ServiceId?
`we found for NRK P1+          serviceId f209`

JvanKatwijk, 2017-02-23:
I think you mean a "challenge" more than a problem. I think it is a fair
suggestion, I'll put it on the "todo" list
j

JvanKatwijk, 2017-02-23:
It was raining and storming today, so I stayed in.

I am not sure I had the separated version in git. Anyway, the basic idea is
to have a separate library with the dab handling
and some interface, just using that library

To "run" the example you must create the library (goto dab-library), and
install it
then you goto the example and build it.

There is in the example a "secret" switch, the -S command line switch. If
you pass a parameter - i.e. a serviceId in hex - then
- if that serviceId is to be found - that will run.

All highly experimental, but is works nice
best
jan

Razmuzh, 2017-02-23:
Beautiful! Thanks!
Will compile and test tonight.

best
R.

Razmuzh, 2017-03-03:
Works perfect!
Many thanks!

closed 2017-03-03 by Razmuzh