JvanKatwijk/dab-cmdline #5: i2c wr failed

rickwookie, 2017-03-16:
Had a bunch of issues getting this to compile and run on a clean RPi 3 raspbian install (Linux raspberrypi 4.4.26-v7+ #915 SMP Thu Oct 20 17:08:44 BST 2016 armv7l GNU/Linux), but got there in the end.

Anyway, trying to use this it seams that there's a problem communicating with the dabstick hardware:
...
Found Rafael Micro R820T tuner
samplerate set to 2048000
Supported gain values (29): r82xx_write: i2c wr failed=-9 reg=09 len=1
r82xx_set_freq: failed=-9
going to tell the world
no ensemble, quittingwe told the world
ofdmProcessor is shutting down

Any ideas where to start with this?

JvanKatwijk, 2017-03-16:
r82xx_write: i2c wr failed=-9 reg=09 len=1

Are you sure that the stick is supported by the osmocom driver?
best
jan

rickwookie, 2017-03-16:
ok so I just installed rtl-sdr, the ran rtl_test -t and get:
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
Sampling at 2048000 S/s.
No E4000 tuner found, aborting.

Which I think is normal right?

Then I did rtl_fm -M wbfm -f 92.6M | aplay -r 32k -f S16_LE -t raw -c 1
and I get:
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to automatic.
Tuned to 92871000 Hz.
Oversampling input by: 6x.
Oversampling output by: 1x.
Buffer size: 8.03ms
Exact sample rate is: 1020000.026345 Hz
Sampling at 1020000 S/s.
Output at 170000 Hz.
Playing raw data 'stdin' : Signed 16 bit Little Endian, Rate 32000 Hz, Mono

...and from the headphone jack on the RPi3 I can hear the radio station.

rickwookie, 2017-03-16:
For completeness, here's the full output I'm getting from dab_cmdline:

pi@raspberrypi:~/dab-cmdline/example/build $ ./dab_cmdline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
Hostapis: 2
Api 0 is ALSA
Api 1 is OSS
selecting device default
device bcm2835 ALSA: - (hw:0,0) seems available as 0
device bcm2835 ALSA: IEC958/HDMI (hw:0,1) seems available as 1
device sysdefault seems available as 2
device default seems available as 3
device dmix seems available as 4
Suggested size for outputbuffer = 512
OK, functions seem to be loaded
Found Rafael Micro R820T tuner
samplerate set to 2048000
Supported gain values (29): r82xx_write: i2c wr failed=-9 reg=10 len=1
r82xx_set_freq: failed=-9
going to tell the world
no ensemble, quittingwe told the world
ofdmProcessor is shutting down

JvanKatwijk, 2017-03-16:
I am not able to reconstruct the behaviour you are describing, that makes
it more complex to "debug".
The error line is
Supported gain values (29): r82xx_write: i2c wr failed=-9 reg=10 len=1
The expected output here would be as stated in your previous mail.

Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6
19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9
44.5 48.0 49.6

So I can precisely tell you in which line of the code it goes wrong, but
since it happens in a function contained in the rtlsdr library.
so I have to dig somewhat deeper

Jan

rickwookie, 2017-03-16:
Thanks.

Further info, my versions of librtlsdr and libusb:
pi@raspberrypi:~ $ dpkg -s librtlsdr0 librtlsdr-dev libusb-1.0-0 libusb-1.0-0-dev | grep Version
Version: 0.5.3-3
Version: 0.5.3-3
Version: 2:1.0.19-1
Version: 2:1.0.19-1

hope that's useful.

JvanKatwijk, 2017-03-16:
Is there difference in behaviour if you run the program as root?
I never us the rtlsdr package that is part of the distribution, I download
recent sources and build a version
jan

rickwookie, 2017-03-16:
Same error when run as root.

Could you maybe point me in the direction of the version of librtlsdr that you're using?

JvanKatwijk, 2017-03-16:
I am always downloading the rtlsdr lib from the osmocom site
I contains a description how to compile and install (
http://osmocom.org/projects/sdr/wiki/Rtl-sdr)
Jan

rickwookie, 2017-03-16:
Hmmm there's a suggestion here: http://www.hamradioscience.com/raspberry-pi-as-remote-server-for-rtl2832u-sdr/#comment-27496 that it could be power related, but I've tried a few power supplies and rtl_test never fails on me with that error, only dab_cmdline. Also I can listen to FM radio using rtl_fm with no issues. It's not likely that a power issue would occur ONLY when running dab_cmdline is it?

I'll try to build rtl-sdr from source. Does that create librtlsdr-dev too?

rickwookie, 2017-03-16:
I've removed the repo packages librtlsdr0 librtlsdr-dev rtl-sdr and built rtl-sdr from source as you've suggested. The only difference now is an extra line in the output of both rtl_test and dab_cmdline:
[R82XX] PLL not locked!

pi@raspberrypi:~/dab-cmdline/example/linux-bin $ rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
^CSignal caught, exiting!

User cancel, exiting...
Samples per million lost (minimum): 0

pi@raspberrypi:~/dab-cmdline/example/linux-bin $ ./dab_cmdline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
Hostapis: 2
Api 0 is ALSA
Api 1 is OSS
selecting device default
device bcm2835 ALSA: - (hw:0,0) seems available as 0
device bcm2835 ALSA: IEC958/HDMI (hw:0,1) seems available as 1
device sysdefault seems available as 2
device default seems available as 3
device dmix seems available as 4
Suggested size for outputbuffer = 512
OK, functions seem to be loaded
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
samplerate set to 2048000
Supported gain values (29): r82xx_write: i2c wr failed=-9 reg=09 len=1
r82xx_set_freq: failed=-9
going to tell the world
no ensemble, quittingwe told the world
ofdmProcessor is shutting down

rickwookie, 2017-03-17:
Done a bit more digging and it seems that the RPi 3 could well need a 5V 2.5A PSU. Best I've tried is 5.1V 2.1A. I'm awaiting delivery of a couple of 5V 3A adapters and I'll let you know if this is the issue.

rickwookie, 2017-03-21:
I've finally got DAB!

IT was the shockingly bad DAB signal that was the reason I wasn't getting it before. Managed now using the local ensemble and holding the antenna VERY HIGH.

This output message still persists though, although it seems perhaps it can be ignored if it has no effect on the functioning of the program.

./dab_cmdline -C 10D -P "3" -G 496
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
Hostapis: 2
Api 0 is ALSA
Api 1 is OSS
selecting device default
device bcm2835 ALSA: - (hw:0,0) seems available as 0
device bcm2835 ALSA: IEC958/HDMI (hw:0,1) seems available as 1
device sysdefault seems available as 2
device default seems available as 3
device dmix seems available as 4
Suggested size for outputbuffer = 512
OK, functions seem to be loaded
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
samplerate set to 2048000
Supported gain values (29): r82xx_write: i2c wr failed=-9 reg=09 len=1
r82xx_set_freq: failed=-9
program (1): C8CC Smooth 4Counties in the list
ensemble Herts Beds Bucks
program (2): C363 Heart 4 Counties in the list
program (3): C8DA Encore Radio in the list
program (4): C3CE BOB HomeCounties in the list
program (5): C1CE Connect FM in the list
program (6): C0CE MKFM in the list
program (7): C637 BBC 3CR in the list
program (8): C47F Capital UK in the list
going to tell the world
we found for BBC 3CR serviceId c637
selected BBC 3CR
we told the world
we have now DAB

JvanKatwijk, 2017-03-21:
Looks interesting, I wonder then where the error messages are coming from.
One would suspect that the frequency is not set prior to the samplerate
setting (the thing reports that the PLL is not locked).

I'll look into the rtl-sdr code
best
jan

MkMunich, 2017-04-04:
Hi,
I've the same issue with my Astrometa DVB-T2 stick (ID 15f4:0131 HanfTek). The stick was not found by ./dab_cmdline when using librtlsdr from wget http://sm5bsz.com/linuxdsp/hware/rtlsdr/rtl-sdr-linrad4.tbz

After having installed rtlsdr from git clone git://git.osmocom.org/rtl-sdr.git, the stick was found by dab_cmdline, but the error below is given:
pi@raspberrypi:~/dab-cmdline/example/build $ ./dab_cmdline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

connect(2) call to /tmp/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
Hostapis: 2
Api 0 is ALSA
Api 1 is OSS
selecting device default
device bcm2835 ALSA: - (hw:0,0) seems available as 0
device bcm2835 ALSA: IEC958/HDMI (hw:0,1) seems available as 1
device sysdefault seems available as 2
device default seems available as 3
device dmix seems available as 4
Suggested size for outputbuffer = 512
OK, functions seem to be loaded
Found Rafael Micro R828D tuner
[R82XX] PLL not locked!
samplerate set to 2048000
Supported gain values (29): r82xx_write: i2c wr failed=-9 reg=09 len=1
r82xx_set_freq: failed=-9
going to tell the world
no ensemble, quittingwe told the world
ofdmProcessor is shutting down

I'm using a Rpi3 and Debian Jessie. Confusing me: why is "apt-get install ... rtl-sdr ..." necessary in step 1 of qt-dab project, when in step 2 the rtl-sdr-linrad4 is build and installed??

Do you need more information?

JvanKatwijk, 2017-04-04:
Seems to me that something is double, very well possible. Apparently the
list of supported devices for the osmocom software is
larger than the more specialized - and older - library from linrad.

The error you are getting is seen before, since I cannot reconstruct it
here it is a little difficult to react.  From what I ve seen before,
the stick functions, regardless of the error message -9, and if it
functions it sends data.
Apparently no ensemble was received, since that is what the report is
telling : no ensemble found.

The difficulty with command line versions is that you do not see anything,
no signal no constellation so there is no way to
tell whether the signal is decent or not.

MkMunich, 2017-04-04:
A bit output is provided :-)
the issue is, that the default values do (normally) not 'point' to a valid radio station. Having this in mind, I connected my stick to Windows and ran qt-dab-0.998-s.exe first, in order to figure out one radio station, which I then provided to dab_cmdline on RPi 3 side. Here a lucky result in my case:
    pi@raspberrypi:~/dab-cmdline/example/build $ ./dab_cmdline -C 5C -P "Deutschlandfunk" -G 90
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

connect(2) call to /tmp/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
Hostapis: 2
Api 0 is ALSA
Api 1 is OSS
selecting device default
device bcm2835 ALSA: - (hw:0,0) seems available as 0
device bcm2835 ALSA: IEC958/HDMI (hw:0,1) seems available as 1
device sysdefault seems available as 2
device default seems available as 3
device dmix seems available as 4
Suggested size for outputbuffer = 512
OK, functions seem to be loaded
Found Rafael Micro R828D tuner
[R82XX] PLL not locked!
samplerate set to 2048000
Supported gain values (29): r82xx_write: i2c wr failed=-9 reg=09 len=1
r82xx_set_freq: failed=-9
program (1): 15DD RADIO BOB!       in the list
program (2): 17FA Absolut relax    in the list
program (3): 10C3 SCHLAGERPARADIES in the list
program (4): 1A45 ENERGY           in the list
program (5): D210 Deutschlandfunk  in the list
program (6): 100D Schwarzwaldradio in the list
program (7): D240 DRadio DokDeb    in the list
program (8): 1A64 ERF Plus         in the list
ensemble DR Deutschland
program (9): D220 DKultur          in the list
program (10): D01C Radio Horeb      in the list
program (11): 15DC sunshine live    in the list
program (12): D75B KLASSIK RADIO    in the list
program (13): D230 DWissen          in the list
going to tell the world
we found for Deutschlandfunk  serviceId d210
selected Deutschlandfunk
we told the world
we have now DAB+

It looks like, dab_cmdline is performing a scan using the given channel (e.g. "-C 5C"). The gain value seems to be important. Audio is played using the default ALSA device.

BUT... unfortunatelly I have to execute dab_cmdline very often (>50 times) to get it working like above. All the other tries are not successful.

I'm not the expert and I'm just thinking loud... Could it be, that the detection of a valid starting point of the stream could be improved? Is 'Autogain' supported, that I do not need to care about it?

JvanKatwijk, 2017-04-04:
The gain value is extremely important. Note however that - from an
implementation point of view - the qt-dab and dab-cmdline are equivalent. I
do not know the receiving conditions in your area, but it is my experience
that using the settings as seen optimal for the qt-dab are also working
good for the dab-cmdline version.

I'll think on the autogain, might be a good suggestion

best
jan

JvanKatwijk, 2017-04-04:
I added an entry in the dab-api for setting (or unsetting) the autogain.
The cmdline example (the C++) version now has an option 'Q' to set autogain

MkMunich, 2017-04-05:
Hi Jan,
to give you a feedback: I tried the latest sources and after rebuilding I tested the command:
   ./dab_cmdline -C 5C -P "Deutschlandfunk" -Q
It does not play. Error: no ensemble, ...
Where as when ommiting option -Q it is working (probably, because of a default gain value?).

andimik, 2017-04-06:
Then I would say: Use another stick. Nowadays people are throwing away or at least selling them as there are no DVB-T signals in lots of regions any more (Munich included). Try to get one with E4000 tuner, which were quite rare. NOXON-Stick v3 might be the right one but nobody would sell it.

I have 6 different sticks at home.

JvanKatwijk, 2017-04-06:
I do not know much about the sticks, I seldom use one, but is the autogain
value dependent on the reglar gain value that is set?
If so, use both gain and autogain since in the implementation first the
gain is set then the autogain'

MkMunich, 2017-04-06:
@andimik I tried a NooElec NESDR SMArt  (having a R820T2 tuner IC, as I can not help with E4000) with command:
    ./dab_cmdline -C 11D -P "BR-KLASSIK" -Q
and this stick is working (!!!) even the same error is displayed:
Supported gain values (29): r82xx_write: i2c wr failed=-9 reg=09 len=1
r82xx_set_freq: failed=-9
r82xx_write: i2c wr failed=-9 reg=07 len=1

@JvanKatwijk I just saw sth strange with the command above and using my Astromety stick again:
Supported gain values (29): r82xx_write: i2c wr failed=-9 reg=09 len=1
r82xx_set_freq: failed=-9
r82xx_write: i2c wr failed=-9 reg=07 len=1
program (1): D314 BR-KLASSIK       in the list
program (2): D313 BAYERN 3         in the list
program (3): D911 Bayern 1 Schw    in the list
program (4): D412 Bayern 2 Sued    in the list
going to tell the world
no ensemble, quittingwe told the world
ofdmProcessor is shutting down

Is strange to me, as BR-KLASSIK is in the list, but not found, then?

I also tried:
   ./dab_cmdline -C 11D -P "BR-KLASSIK" -Q # not playing
   ./dab_cmdline -C 11D -P "BR-KLASSIK" -G 80 # playing
   ./dab_cmdline -C 11D -P "BR-KLASSIK" -G 80 -Q # not playing

If I understood you right, then the third command should play radio?

JvanKatwijk, 2017-04-06:
The confusion may arise that you see program names passing by. However, the
decision whether or not an ensemble is found ismade by having recognized -
or not - the name of the ensemble. If that happens, then it is assumed that
the synchronization has reached its final offset and is further switched
off. That is just a choice.

Since the software (at least the qt-dab software) should run on a rpi 2,
and since the synchronization as such is pretty heave, synchronization
takes two threads to execute, and they obviously have to talk to each
other. In my experience, switching sync off on the ensemble name is more
reliable than switching off on a program name

Wrt the last question: I do not know wat autogain is doing related to the
regular gain setting, all I am saying is that the current implementation
first does the regular gain setting, then the switches the autogain on (my
stick in mechanically extremely weak, and sdrplay and airspy outperform a
stick to such an extent that I only use the stick for specific tests)

andimik, 2017-04-06:
It is not the best idea to start testing a stick with this command line tool. Please try qt-dab or welle.io first, then you will see, which gain parameter is the best for your area, your stick and your hardware. 

Every stick is different and supports lots of gain settings. In your case 29.

MkMunich, 2017-04-06:
Dear andimik, thanks for your hint - I know that both sticks are working in qt-dab and in dab-cmdline, when using the correct gain values + other settings. Perhaps you did not know, that when using in a driving car an 'Autogain' function is not just nice to have?

JvanKatwijk, 2017-04-06:
The autogain function works pretty well with an sdrplay device, so the
first thing that we have to find out is what the combination autogain on
and gain setting does - and can be influenced - for simple sticks. The
sensitivity of the sticks is obviously limited and they deliver only 8
bits, one sign bit and some noise and you have a or 4 bit left which does
not deliver enoughto decode

I'll dig up my stick and run a few tests with qt-dab to see what the
behaviour is with gainsetting/autogain

JvanKatwijk, 2017-04-06:
It turns out that after setting autogain "on", you better set a gain value,
so I made a change in that direction on the dabstick handling software.

MkMunich, 2017-04-06:
Hi Jan, I've a short question: I saw, that the configuration could be set from outside. This is a very good idea!
But I do not know, how to call it, so I commented out the if statement in CMakeLists.txt like:
#Mk if(DEFINED DABSTICK)
   set(DABSTICK true)
#Mk endif ()

Could you give an example, how to call the build of e.g. DABSTICK ?

MkMunich, 2017-04-06:
Hi Jan, your update is looking ok. the command:
    ./dab_cmdline -C 11D -P "BR-KLASSIK" -G 80 -Q
is now working. This is what I thought, that the order was wrong before.
GREAT!

The error:
Supported gain values (29): r82xx_write: i2c wr failed=-9 reg=09 len=1
r82xx_set_freq: failed=-9
r82xx_write: i2c wr failed=-9 reg=07 len=1
is still present, but not causing damages. I did not dig into it, and I'm only guessing, that it might be a wrong access to the gain value table? This would be a typical error from myself ;-)

JvanKatwijk, 2017-04-06:
the cmake command understands -DDABSTICK=ON, so assuming you created a
build directory
cmake .. -DDABSTICK=ON

JvanKatwijk, 2017-04-06:
Good to hear, tomorrow I am going to play with the dab-cmdline and the
dabstick. Let you know the results, more people are complaining about the
error you mentioned

best
jan

MkMunich, 2017-04-07:
Wow Jan, you're so quick :-)
The latest sources solved the i2c wr issue.

There are 2 comments to build dab-cmdline:
1) Is there a need for python 3.5.*? The installed python version in Jessie is 2.7.x. I got 3.4 with:
    sudo apt-get update
    sudo apt-get install python3-dev
Then I had a need to adjust all paths to point to '/usr/include/python3.4m' in ~/dab-cmdline/dab-library/CMakeLists.txt

2) There is a small dependency issue in the build order, because of the new default, that PYTHON is set in
~/dab-cmdline/dab-library/CMakeLists.txt. The quick&dirty solution for me was to comment line:
    #  set(PYTHON true)
 And built using the following commands (in my case!):
    cmake .. -DDABSTICK=ON
    make -j4
    sudo make install
Then I deleted the '#' and did the same commands again.

Have a nice weekend.

JvanKatwijk, 2017-04-07:
Oops, I was playing this afternoon and tested with Python as well, of
course set Python had to be commented out
best
jan

rickwookie, 2017-04-10:
Well, the i2C error has indeed gone with the new build (not sure what you did to fix that), but now it segfaults every time after about 1 second of audio, with or without the autogain flag:

pi@raspberrypi:~/dab-cmdline/example/build $ ./dab_cmdline -C 10D -P "BBC" -G 100 -Q
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
Hostapis: 2
Api 0 is ALSA
Api 1 is OSS
selecting device default
device bcm2835 ALSA: - (hw:0,0) seems available as 0
device bcm2835 ALSA: IEC958/HDMI (hw:0,1) seems available as 1
device sysdefault seems available as 2
device default seems available as 3
device dmix seems available as 4
Suggested size for outputbuffer = 512
OK, functions seem to be loaded
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
samplerate set to 2048000
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
input device is rtlsdr handler ? 1
ensemble Herts Beds Bucks
waited for an ensemble, an ensemble is found
Encore Radio - with SId C8DA - in ensemble
BOB HomeCounties - with SId C3CE - in ensemble
MKFM - with SId C0CE - in ensemble
Connect FM - with SId C1CE - in ensemble
Capital UK - with SId C47F - in ensemble
BBC 3CR - with SId C637 - in ensemble
Heart 4 Counties - with SId C363 - in ensemble
Smooth 4Counties - with SId C8CC - in ensemble
we found for BBC 3CR serviceId c637
selected BBC 3CR
we have now DAB
Segmentation fault

JvanKatwijk, 2017-04-10:
Can you run with gdb and give a stack trace

rickwookie, 2017-04-10:
Starting program: /home/pi/dab-cmdline/example/build/dab_cmdline -C 10D -P "BBC" -G 100 -Q
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
[New Thread 0x7644c450 (LWP 1477)]
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
Hostapis: 2
Api 0 is ALSA
Api 1 is OSS
selecting device default
[Thread 0x7644c450 (LWP 1477) exited]
device bcm2835 ALSA: - (hw:0,0) seems available as 0
device bcm2835 ALSA: IEC958/HDMI (hw:0,1) seems available as 1
device sysdefault seems available as 2
device default seems available as 3
device dmix seems available as 4
Suggested size for outputbuffer = 512
[New Thread 0x761ff450 (LWP 1478)]
OK, functions seem to be loaded
[New Thread 0x759ff450 (LWP 1479)]
[Thread 0x759ff450 (LWP 1479) exited]
[New Thread 0x759ff450 (LWP 1480)]
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
samplerate set to 2048000
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
input device is rtlsdr handler ? 1
[New Thread 0x73ee2450 (LWP 1481)]
[New Thread 0x736e2450 (LWP 1482)]
[New Thread 0x72ee2450 (LWP 1483)]
[New Thread 0x722ff450 (LWP 1484)]
ensemble Herts Beds Bucks
waited for an ensemble, an ensemble is found
Encore Radio - with SId C8DA - in ensemble
BOB HomeCounties - with SId C3CE - in ensemble
MKFM - with SId C0CE - in ensemble
Connect FM - with SId C1CE - in ensemble
Capital UK - with SId C47F - in ensemble
BBC 3CR - with SId C637 - in ensemble
Heart 4 Counties - with SId C363 - in ensemble
Smooth 4Counties - with SId C8CC - in ensemble
we found for BBC 3CR serviceId c637
selected BBC 3CR
we have now DAB
[New Thread 0x712fe450 (LWP 1485)]

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x712fe450 (LWP 1485)]
0x00000000 in ?? ()

JvanKatwijk, 2017-04-10:
with bt, when in gdb you get a stacktrace, which is pretty helpful in
deciding where things go wrong (next to me I have a running version without
problem)

closed 2018-02-12 by JvanKatwijk