JvanKatwijk/dab-cmdline #6: Cannot compile, CMakeLists.txt not found

andimik, 2017-03-16:
> andreas@T42:~$ cd dab-cmdline/
> andreas@T42:~/dab-cmdline$ mkdir build
> andreas@T42:~/dab-cmdline$ cd build/
> andreas@T42:~/dab-cmdline/build$ cmake ..
> CMake Error: The source directory "/home/andreas/dab-cmdline" does not appear to contain CMakeLists.txt.

I followed your suggestions, but I could not compile it. I found the file there:

> ./example/CMakeLists.txt
> ./dab-library/devices/rtl_tcp/CMakeLists.txt
> ./dab-library/CMakeLists.txt

Am I wrong?

JvanKatwijk, 2017-03-17:
It seemed obvious to be that when compiling the dab library, one should
"cd" to that library, and when compiling the example program one should
"cd" to the example directory. I made it more explicit now in the README
text

JvanKatwijk, 2017-03-30:
> What does this sentence mean (sorry for that stupid beginner question)?
>
> IMPORTANT: YOU NEED C++11 SUPPORT FOR THIS
>
> I cannot compile it.
>
> #error This file requires compiler and library support for the ISO C++
> 2011 standard. This support must be enabled with the -std=c++11 or
> -std=gnu++11 compiler options.
>
> I found lots of suggestions to enable c++11 in qt-creator, but they all
> are for adding a parameter to the pro-file, which does not exist for this
> program.
>
> dpkg -l cpp
> ii cpp 4:5.3.1-1ubu amd64 GNU C preprocessor (cpp)
>
> dpkg -l gcc
> ii gcc 4:5.3.1-1ubu amd64 GNU C compiler
>
> This is Ubuntu 16.04 LTS.

C++11 support is required for the command line versions, since I am using a
lot of C++11 stuff that was provided for by Qt.
The command line version are not compiled by qmake since they do not
contain any Qt stuff.

Nothing has to be done, however, since the indication that C++11 support is
needed is already in the command line options as specified in the CMakeText
files.

Standard gnu compilers spport g++11 most of them implicitly (some older
ones need the command line option to initiate the support)

jan

andimik, 2017-03-30:
Aaaaaa

Now it's clearer :-D

andimik, 2017-04-02:
Sorry, but I can still not compile it.

My steps are:

> cd dab-cmdline/
cd dab-library/
mkdir build
cd build/
cmake -std=gnu++11 .. -DAIRSPY=ON -DDABSTICK=ON
make
sudo make install

(no errors so far) 

and then

> cd ..
> cd example/
> mkdir build
> cd build/
> cmake -std=gnu++11 ..
> make

I tried `cmake -std=gnu++11 ..` without success. Still the same error.

How to enable C++11 in Ubuntu and cmake? 
What is the syntax? 
Or do I have to edit the CMakeText.txt anyway (although you say it's not necessary)?

JvanKatwijk, 2017-04-02:
What is the message you get?

andimik, 2017-04-02:
> #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.

JvanKatwijk, 2017-04-02:
Did you try to add -std=c++11 ?  The CMakeLists.txt file has a line
add_definitions ( -Wall -g -std=c++11)

So, apparently that worked. if adding to the command line does not work,
then yo can add -std=c++11 in the CMakeLists.txt file.
Anyway, I'll add it and upload it to github

andimik, 2017-04-02:
Yes, when I add it to the CMakeLists text file then it works.

Same with ETI stuff (cmdline). Now I could build it as well. :-)

JvanKatwijk, 2017-04-02:
I added it to the CMakeLists file in dab-cmdline, and I'll add it to the
CMakeLists file in eti-stuff

closed 2017-04-02 by andimik