JvanKatwijk/dab-cmdline #8: Get complete ensemble via ETI or API

basicmaster, 2017-03-20:
It would be great to have an option to get the complete ensemble as ETI, instead of just a particular channel (maybe `eti-frontend` of your `eti-stuff` would be a better starting point for this feature; currently it outputs already ETI, but needs further user interaction).

Or to just get the FIC and any desired subchannel in "raw mode" via the API.

This way one would have a replacement for `dab2eti`of the `dabtools` when using DABlin and would be able to easily use other hardware than a Realtek stick with it.

JvanKatwijk, 2017-03-20:
Hi

Can you be more specific? afaik the ensemble on a selected channel is
complete in the eti file

basicmaster, 2017-03-20:
What I am talking about is a tool, that does not need user interaction (all relevant params via cmdline) and outputs ETI of a specific DAB channel. Hence kind of combination of `dab-cmdline` and `eti-frontend` would be great:
- `dab-cmdline` is completely controlled via cmdline and does not need user interaction, but does not support ETI output; instead one has to select a specific station
- `eti-frontend` has ETI output, but the user has to perform some clicks to start the ETI output

So something like `dab2eti`, but `dab2eti` itself has unfortunately some flaws, like only support for Realtek sticks.

russel, 2017-03-21:
I was still my intention to rewrite dab2eti so as to make it better in the dablin context. I particularly wanted to support /dev/swradio0 as well as librtlsdr. I have only Realtek devices so that limits my scope for supporting more. A backdrop here is libdvbv5 which acts as middleware between the /dev/dvb/* and applications. Making dab2eti (or whatever it is called) an effective middleware layer between devices or other libraries and applications. Of course whilst libdvbv5 is just a classic library, dab2eti is a standalone process in good modern microservices, dataflow architecture (or is this laying it on a bit thick?).

If there is a wider context in which such a program would be useful then I am certainly "up for that" as the saying goes.

cf.  https://github.com/Opendigitalradio/dabtools/issues/12

The discussion there was centred on whether to try Rust or D as an implementation language rather than sticking with C++11, (switching to C++14 or C++17 was thought perhaps a little risky) but that was in the dablin context. If there is a wider context, the issues change.

Razmuzh, 2017-03-24:
Is it possible to omit the ensemble list, and just connect to the channel - Switching channels take a lot of time when the whole list has to be generated every time.

JvanKatwijk, 2017-03-27:
That is certainly possible. I'll think about it. It is on the todo list
best
jan

JvanKatwijk, 2017-03-29:
I added a first version of a command line version (same combination,
library + a main program) to the eti-stuff repository.

As with dab cmdline, the supported device is compile time selected.

Jan

2017-03-27 22:02 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> That is certainly possible. I'll think about it. It is on the todo list
> best
> jan
>
>
>
>> What I am talking about is a tool, that does not need user interaction
>> (all relevant params via cmdline) and outputs ETI of a specific DAB
>> channel. Hence kind of combination of dab-cmdline and eti-frontend would
>> be great:
>>
>>    - dab-cmdline is completely controlled via cmdline and does not need
>>    user interaction, but does not support ETI output; instead one has to
>>    select a specific station
>>    - eti-frontend has ETI output, but the user has to perform some
>>    clicks to start the ETI output
>>
>> So something like dab2eti, but dab2eti itself has unfortunately some
>> flaws, like only support for Realtek sticks.
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

basicmaster, 2017-04-22:
Thanks, Jan!

closed 2017-04-22 by basicmaster