JvanKatwijk/dab-cmdline #10: "it appeared that you did not select a device"

andreacampanella, 2017-04-25:
I'm running dab_cmdline on a raspberry pi 2 but I'm receiving this message : 

"it appeared that you did not select a device, 
you can run, but you will be connected to a virtual device, only providing zeros"

my lsusb

Bus 001 Device 005: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T

the usb stick works properly with rtl_fm , I have no idea how to solve this honestly...

JvanKatwijk, 2017-04-25:
Wel, it seemed you did not select a device. Did you add something like ..
-DDABSTICK=ON to the cmke command?

andreacampanella, 2017-04-26:
oh, yes, that was it, now it recognise the stick but it can't find an ensemble: 
"waited for an ensemble, an ensemble is not found
no ensemble, quittingofdmProcessor is shutting down"
any idea?

JvanKatwijk, 2017-04-26:
It is pretty tough without any further indication to locate the ensemble.
What I normally do is to set the gain to a value that I have seen working
with qt-dab

closed 2018-02-12 by JvanKatwijk