JvanKatwijk/dab-cmdline #12: Cmake error compiling the python example

andreacampanella, 2017-06-13:
I'm really confused about the way I should compile the python example, I'm following the README ,but the cmake ( cmake .. -DSDRPLAY=ON) command generate a rather cryptic error :

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
SDRPLAYLIB_INCLUDE_DIR
   used as include directory in directory /home/andrea/dab-cmdline/python-example
   used as include directory in directory /home/andrea/dab-cmdline/python-example

andreacampanella, 2017-06-15:
Ok, I managed to compile this , I was using the wrong define, using -DRTLSDR I can compile it. 
but now I have another problem , if I run python3 cmdline.py  -d 5 -c 12A -p "Fix Radio" 

I get : 

device restarted
ofdm word gestart
no DAB signal detected, giving up
no ensemble found, giving up

I had to choose python3.4m instead of python3.5m to make it working on the raspberry pi .

I'm trying to make a digital radio receiver without a screen attached, I'm really finding difficult, there is not a single working or ready piece of software to use the raspberry  as DAB receiver via command line, I can't get it...

andimik, 2017-06-15:
Have you tried eti-stuff (including eti-cmdline)  as well?

This is much easier and stable.

andreacampanella, 2017-06-15:
hey andimik , thank! 
Do I have to feed the eti-cmd output in the input of something else or is standalone?

JvanKatwijk, 2017-07-04:
Hi
I was on (an internet free) holiday, so actually I did not read mail for
app 4 weeks. I really can recommend that.
Wrt your issue: What you get (your mail) is just that it works fine.
probably the gain has to be set differently

In the same environment as the Python example you'll find a well working
command line version.

best
jan

jehutting, 2017-10-15:
THANKS a lot @andreacampanella.

I was struggling to get this library/example-1 working for the exact same purpose as you do: having a stand-alone -without screen- DAB receiver/player. I was using the wrong define too.
 
Although the python-example still fails
```
$ python3 cmdline.py 
device restarted
ofdm word gestart
no DAB signal detected, giving up
no ensemble found, giving up
```
_(and installing python3.4 gives me import errors as numpy/multiarrays/... and I don't know how all this relates to python3.5, the latest and greatest on the Raspbian Stretch)_

I'm now able to compile example-1 (after I patched it...
```
$ git diff
diff --git a/example-1/main.cpp b/example-1/main.cpp
index 010f85b..aa83ddd 100644
--- a/example-1/main.cpp
+++ b/example-1/main.cpp
@@ -115,10 +115,11 @@ void      dataOut_Handler (std::string dynamicLabel, void *ctx) {
 }
 
 static
-void   bytesOut_Handler (uint8_t *data, int16_t amount, void *ctx) {
+void   bytesOut_Handler (uint8_t *data, int16_t amount, uint8_t type, void *ctx) {
        (void)data;
        (void)amount;
        (void)ctx;
+       (void)type;
 }
 //
 //     The function is called from within the library with
```
).

So, ```cmake .. -DRTLSDR=ON``` (in the created build directory of example-1) followed by `make`, gives `dab-rtlsdr`. _Note: during my struggle I installed/removed a lot of packages; hang-on while building/compiling :-)_

For example
```
$ ./dab-rtlsdr -C 12C -P "NPO Radio 2"
dab_cmdline, 	                  Copyright 2017 J van Katwijk, Lazy Chair Computing
OK, functions seem to be loaded
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
samplerate set to 2048000
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
[... 61 lines cut ...]
NPO SterrenNL    (8212) is part of the ensemble
NPO Radio 4      (8204) is part of the ensemble
NPO 3FM          (8203) is part of the ensemble
NPO Radio 5      (8205) is part of the ensemble
NPO 3FM KX Radio (8214) is part of the ensemble
NPO Radio 4 Con. (8215) is part of the ensemble
ensemble NPO              is (8001) recognized
NPO Radio 1      (8201) is part of the ensemble
NPO FunX Dance   (8216) is part of the ensemble
NPO 3FM Alterna. (8210) is part of the ensemble
NPO Radio 2      (8202) is part of the ensemble

going to start program NPO Radio 2
selected NPO Radio 2
	startaddress	= 0
	length		= 72
	subChId		= 1
	protection	= 2
	bitrate		= 96
we have now DAB+

```
I don't know (yet) what ALSA complains about, but it just works!!!

@JvanKatwijk "Uiteraard een grote DANKUWEL voor de library en het voorbeeld programma!".

JvanKatwijk, 2017-10-23:
apologies for the late reply. I was/am in the process of changing from one
laptop to another and - while the new laptop as such is an improvement -
the process
is a nightmare.

It is not easy to get the software up. It took actually quite a while
before the python example was running on my new laptop.
Do not nother about the complaints of alsa, I do not think anyone
understands that. The important thing is that
sound is coming out.

Your first example merely shows that no reasonable signal could be
identified. Your second example
makes me feel good.

Thanks for the comment
best
jan

closed 2018-02-12 by JvanKatwijk