JvanKatwijk/dab-cmdline #13: ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave

sjdean, 2017-07-31:
So far I've not had any problems with Audio, in that PulseAudio works fine for A2DP Bluetooth, Hands Free, local music playback, web browsing, and also rtl_fm.

However both example-1 and example-2 give me the following. Any thoughts?

Im looking to build a CarPC, so hoping I can build up on this. Looking to build a Python front end in KODI and hopefully get a channel scan and tuning setup. Hopefully borrow some bits from your Qt version at some point :-)

First thing though, is getting some audio from the thing!

Simon

pi@master:~/Downloads/dab-cmdline-master/example-1/build $ ./dab_cmdline -M 1 -B "BAND III" -C 12C -P "Radio 4" -G 80 -A default
dab_cmdline, 	                  Copyright 2017 J van Katwijk, Lazy Chair ComputingALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Hostapis: 2
Api 0 is ALSA
Api 1 is OSS
selecting device default
device bcm2835 ALSA: - (hw:0,0) seems available as 0
device bcm2835 ALSA: IEC958/HDMI (hw:0,1) seems available as 1
device sysdefault seems available as 2
device pulse seems available as 3
ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave
.
.
.
ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1735
Expression 'AlsaOpen( hostApi, parameters, streamDir, &pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1767
device default seems available as 5
Suggested size for outputbuffer = 1280
Segmentation fault

JvanKatwijk, 2017-07-31:
It does not look like an output problem. What device did you specify?

sjdean, 2017-07-31:
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

Not quite sure how to specify the device. Tried -A default, bcm2835 and
pulse with same results. The Qt-DAB works fine and that goes to default.

JvanKatwijk, 2017-08-02:
> +31 (0)15 3698980 <+31%2015%20369%208980> <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355> <+31%206%2028260355>
> 319192918>,

No I mean in generating the executable.
mkdir build
cd build
cmake .. -DXXX=ON

The error you showed was more likely an error with the radio device than
with the audio output

closed 2018-02-12 by JvanKatwijk