JvanKatwijk/dab-cmdline #14: Dynamic Label method not quite right

nottledim, 2017-08-20:
For some reason I don't receive any DL messages.  However the dataOut callback does print some garbage characters.  So I was looking at the code and I think there's a misunderstnding.  Either mine or yours :-) I can only find a spec for DL+ so I'm not sure.

Referring to padHandler::dynamicLabel method in pad-handler.cpp
So far as I can tell the DL command field comprises one byte header and field_1 message bytes.  This is why field_1 is specified as one less than DL Command length.

This makes the header length 3 byes not 2 bytes and affects dataLength calculation and the start byte of the message, which should be data[3] not data[2].

Changing this removes the garbage data problem I was seeing but as I receive no real messages I can't test it any further.

c.f. http://www.etsi.org/deliver/etsi_ts/102900_102999/102980/01.01.01_60/ts_102980v010101p.pdf

JvanKatwijk, 2017-08-21:
> 01.01.01_60/ts_102980v010101p.pdf

Thanks for the info. I'll have a look this afternoon

nottledim, 2017-08-21:
Actually, I think I've got it wrong. It's all to do with Cflag. DL+ is 
carried as a DL command so CFlag is 1 whereas your code is for Cflag =0.

It begs the question why I'm receiving garbage characters.

Thanks

Dick

JvanKatwijk, 2017-08-21:
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>
> 323664633

What might be the case is that the test on the CFlag is ONLY done for the
first segment, it seems that for successor segments
no C flag is tested. Since I never was confronted with a Cflag 1 that did
not harm apparently.

Looking at the code, it seems that whenever the C flag is on, the behaviour
is undefined and depends (a.o) on the value of moreXpad

To me it seems that an explicit check on the CFlag is needed.
best
jan

JvanKatwijk, 2017-08-21:
Interesting problem. Actually I have no clue.
Is it possible for you to record a fragment of raw input data that shows
the problem and send me a link?
Otherwise, pls add some print statements, there are e.g. two occurrences of
showLabel

best
jan

2017-08-21 15:05 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> Forget the previous mail. I have to look into what I have written down
> there (just cannot follow it anymore)
> j
>
> 2017-08-21 14:04 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:
>
>> What might be the case is that the test on the CFlag is ONLY done for the
>> first segment, it seems that for successor segments
>> no C flag is tested. Since I never was confronted with a Cflag 1 that did
>> not harm apparently.
>>
>> Looking at the code, it seems that whenever the C flag is on, the
>> behaviour is undefined and depends (a.o) on the value of moreXpad
>>
>> To me it seems that an explicit check on the CFlag is needed.
>> best
>> jan
>>
>>
>>
>>
>>
>>> Actually, I think I've got it wrong. It's all to do with Cflag. DL+ is
>>> carried as a DL command so CFlag is 1 whereas your code is for Cflag =0.
>>>
>>> It begs the question why I'm receiving garbage characters.
>>>
>>> Thanks
>>>
>>> Dick
>>>
>>>
>>>
>>>
>>> > Thanks for the info. I'll have a look this afternoon
>>> >
>>> >
>>> >
>>> >> For some reason I don't receive any DL messages. However the dataOut
>>> >> callback does print some garbage characters. So I was looking at the
>>> code
>>> >> and I think there's a misunderstnding. Either mine or yours :-) I can
>>> only
>>> >> find a spec for DL+ so I'm not sure.
>>> >>
>>> >> Referring to padHandler::dynamicLabel method in pad-handler.cpp
>>> >> So far as I can tell the DL command field comprises one byte header
>>> and
>>> >> field_1 message bytes. This is why field_1 is specified as one less
>>> than DL
>>> >> Command length.
>>> >>
>>> >> This makes the header length 3 byes not 2 bytes and affects dataLength
>>> >> calculation and the start byte of the message, which should be
>>> data[3] not
>>> >> data[2].
>>> >>
>>> >> Changing this removes the garbage data problem I was seeing but as I
>>> >> receive no real messages I can't test it any further.
>>> >>
>>> >> c.f. http://www.etsi.org/deliver/etsi_ts/102900_102999/102980/
>>> >> 01.01.01_60/ts_102980v010101p.pdf
>>> >>
>>> >> <https://github.com/JvanKatwijk/dab-cmdline/issues/14>, or mute the
>>> thread
>>> q5xdJxgAnvrxnQ56AzBWypjks5saKr4gaJpZM4O8vcS>
>>> >>
>>> >
>>> >
>>> >
>>> > Jan van Katwijk
>>> >
>>> >
>>> > +31 (0)15 3698980 <+31%2015%20369%208980>
>>> > +31 (0) 628260355 <+31%206%2028260355>
>>> >
>>> >
>>> > https://github.com/JvanKatwijk/dab-cmdline/issues/14#issueco
>>> mment-323664633
>>>
>>>
>>
>>
>>
>> Jan van Katwijk
>>
>>
>> +31 (0)15 3698980 <+31%2015%20369%208980>
>> +31 (0) 628260355 <+31%206%2028260355>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

nottledim, 2017-08-21:
I can take recordings, but I don't know how.  What should I do?

Dick

JvanKatwijk, 2017-08-22:
> +31 (0)15 3698980 <+31%2015%20369%208980> <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355> <+31%206%2028260355>
> 323689218>,
> 323814422

If you are able to run Qt-DAB, that has a provision to dump the input into
a file. If not, i.e. if you only (can) run the cmdline
version, let me know, then I shall make an extended example-2 version with
a switch to save the input
best
jan

nottledim, 2017-08-22:
I've shared an item with you:

Shared
https://drive.google.com/drive/folders/0B7sIIKn1Z0BPTjFUNXpLVnNmdFk?usp=sharing&ts=599c8482

It's not an attachment – it's stored online. To open this item, just click  
the link above.

This is 1.2G of BBC 6Music transmission on 12B multiplex. I don't know if  
this specifically shows the problem but this program is what I mostly use  
and they always transmit pad (title artist etc).

https://drive.google.com/drive/folders/0B7sIIKn1Z0BPTjFUNXpLVnNmdFk?usp=sharing

nottledim, 2017-08-22:
On 08/22/17 16:54, JvanKatwijk wrote:
> If you are able to run Qt-DAB, that has a provision to dump the input into
> a file. 

There's 1.2G of BBC 6Music here:

https://drive.google.com/open?id=0B7sIIKn1Z0BPcDdkVGp3QXJveUk

I don't know specifically if this shows the problem but I would expect it to
since 6 music transmits program artist title info and I usually what I use
when playing.

Dick

-- 
Dick Middleton
dick@lingbrae.com

JvanKatwijk, 2017-08-23:
Thanks for the link.
Interesting to see that it is all "DAB" rather than DAB+. A quick scan on
my laptop shows that sometimes there is a very brief strange sound (do not
know how to describe it), that suggests that synchronization is lost. I'll
look into it further.

JvanKatwijk, 2017-08-23:
I ran qt-dab both on the laptop and on the rpi2. Result is the same: good
sound, at one point after app minute and a half synchronization
is lost for a very short moment and then a chirp-like sound is heard (which
is caused by the fact that the check on correct MP2 is far
less stringent than the check on correct AAC as is the case with DAB+).

The load on the RPI 2 (controlled headless) is less than 50%. Which brings
me on the following: do you run the RPI with its one
monitor and keyboard and do you have the specrum "on". In that case it
would be good to look at the processor load and the
load on the 4 cores.
best
jan

2017-08-23 15:55 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> Thanks for the link.
> Interesting to see that it is all "DAB" rather than DAB+. A quick scan on
> my laptop shows that sometimes there is a very brief strange sound (do not
> know how to describe it), that suggests that synchronization is lost. I'll
> look into it further.
>
>
>
>> > If you are able to run Qt-DAB, that has a provision to dump the input
>> into
>> > a file.
>>
>> There's 1.2G of BBC 6Music here:
>>
>> https://drive.google.com/open?id=0B7sIIKn1Z0BPcDdkVGp3QXJveUk
>>
>> I don't know specifically if this shows the problem but I would expect it
>> to
>> since 6 music transmits program artist title info and I usually what I use
>> when playing.
>>
>> Dick
>>
>> Dick Middleton
>> dick@lingbrae.com
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

nottledim, 2017-08-23:
On 08/23/17 15:33, JvanKatwijk wrote:
> I ran qt-dab both on the laptop and on the rpi2. Result is the same: good
> sound, at one point after app minute and a half synchronization
> is lost for a very short moment and then a chirp-like sound [...]

Yes, I heard that too.  I don't have a great aerial, or rather it's not in a
good place, so it does lose it sometimes.

The BBC is just DAB.  They were an early adopters and are now stuck with it.

> The load on the RPI 2 (controlled headless) is less than 50%.

I've got an Orange PI One headless.  It's a 4 core H3 processor and it manages
fine.

I've been looking at the PAD messages.  It seems that the BBC sends only short
PAD messages with CI flag = 0.  However the shortPAD decoder seems to me to
not work and I'm busily trying to write something that does.  I have seen
fragments of comprehension in the messages. The short PAD messages should not
be sent to the dynamicLabel routine (if I understand it).

These standards make my head hurt :-)

Dick

--
Dick Middleton
dick@lingbrae.com

nottledim, 2017-08-23:
Jan,

	Here's a patch for pad-handler.  It receives the BBC PAD data and returns a
string to dataOut. I'm not sure how useful it is.

It's not wonderful - it all seems a strange way to go about life.

You'll have to tweak pad-handler-h because I've changed the prototype to shortPAD.

Each string comprises a number of substrings with a max length of 16 chars.
These substrings have 2 trailing chars - I don't know what they mean but I
think they might be some sort of identifier for the line.  They have to be
removed before concatenating the substring into the final string.

Dick

-- 
Dick Middleton
dick@lingbrae.com

JvanKatwijk, 2017-08-24:
Thanks

It seems you forgot the patch itself. Btw yes, the normal way of passing
data is 2 bytes header
and then some characters. The header contains info on whether it is a head,
characterset etc etc

nottledim, 2017-08-24:
On 08/24/17 10:37, JvanKatwijk wrote:
> Thanks
> 
> It seems you forgot the patch itself. Btw yes, the normal way of passing
> data is 2 bytes header
> and then some characters. The header contains info on whether it is a head,
> characterset etc etc

I didn't forget.  It must have been stripped by github.  I've put it on google
drive:

https://drive.google.com/open?id=0B7sIIKn1Z0BPUGExQ1Q4RmhqZDg

It's still not right, I'm missing a trick. Maybe you know what's wrong.
Sometimes messages get concatenated and with ClassicFM the string never gets
sent (so it gets quite big).

There must also be a string identifier of some sort because it keeps repeating
the strings. I read about it somewhere but I can't remember where.

In summary I think the only thing I've achieved is to demonstrate that the BBC
uses shortPAD and the messages can be extracted. It needs to be generalised.

Dick

-- 
Dick Middleton
dick@lingbrae.com

nottledim, 2017-08-24:
On 08/24/17 10:37, JvanKatwijk wrote:

> It seems you forgot the patch itself. Btw yes, the normal way of passing
> data is 2 bytes header
> and then some characters.

It looks like each segment group has two 7 bit characters appended (trailer).
They're either arbitrary or it's some sort of checksum. I can't see any bit
patterns in it.  Do you know anything about that? Quite useful for discarding
repeated messages.

The more I tweak my code the more it looks like your dynamicLabel routine.
Maybe if dynamicLabel could handle the X-PAD subfields (without CI field) it
would save a bit of duplication. Maybe it doesn't matter.

Dick

JvanKatwijk, 2017-08-24:
I made a new version that displays the shortpad dynlables correctly.
What happens is
the AcTy 2 is the start of a new frame of at most 16 characters, the first
character stored within the pad with ActY 2
then a sequence of No CI's is following each with at most 4 characters. The
amount of characters in the frame is
stored in the AcTy 2 pad.
The full message ends with a AcTy 0 pad
pls give it a try
best
jan

nottledim, 2017-08-24:
On 08/24/17 18:23, JvanKatwijk wrote:
> I made a new version that displays the shortpad dynlables correctly.

> The full message ends with a AcTy 0 pad

Yes, that's what I thought.  It works for you but not me??? What happened to
those checksums?

> pls give it a try

It works (almost) and is neater than my version. Well done.

What doesn't work is ClassicFM.  Same problem I found.  What you need to do is
detect first (or last) segment and output the buffer. Classic doesn't seem to
send the AcTy == 0.  Both BBC and Classic send a number of segments so could
do away with AcTy == 0 case.

Dick

JvanKatwijk, 2017-08-24:
I'll have a look (and read the DAB standard) tomorrow
best
jan

JvanKatwijk, 2017-08-25:
Closer look at the data reveals that the AcTy 2 segment contains - next to
the databyte - 3 relevant values,
the length of the text to follow, the segment number and the indication
first/last/intermediate segment

Using the latter now for recognizing the first and last segment of the
messag.
So, no more dependency on AcTy 0

best
jan

nottledim, 2017-08-25:
On 08/25/17 11:05, JvanKatwijk wrote:

> Using the latter now for recognizing the first and last segment of the
> messag.
> So, no more dependency on AcTy 0

Good, except one thing: you need to call dataOut  -  oops :-)

Dick

nottledim, 2017-08-25:
Jan,

	No, sorry, I see what you've done.   The problem was:

                 if ((b [last - 1] & 0xF0) == 0x40) {

should be  if ((b [last - 1] & 0x40) == 0x40) {

Dick

JvanKatwijk, 2017-08-25:
You are right!!! stupid of me. I'll correct it for both the first and last
segment (all those details ... )

nottledim, 2017-08-25:
On 08/25/17 15:22, JvanKatwijk wrote:
> You are right!!! stupid of me. I'll correct it for both the first and last
> segment (all those details ... )

I've cleaned out the cruft and simplified it slightly.  It works just fine on
BBC and ClassicFM.

https://drive.google.com/open?id=0B7sIIKn1Z0BPWkNmN29wQkk1ZmM

Dick

JvanKatwijk, 2017-08-25:
Thanks

firstSegment, lastSegment and still_to_go are now class variables rather
than statics, and there is no need for the data array anymore.
best
jan

nottledim, 2017-08-25:
On 08/25/17 15:47, JvanKatwijk wrote:

> firstSegment, lastSegment and still_to_go are now class variables rather
> than statics, and there is no need for the data array anymore.
> best

Should  toStringUsingCharset be applied to the message?  I assume the CharSet
is in the prefix as in dynamicLabel.

Also I've noticed incomplete message when changing channels.  Probably because
the first segment received isn't the first segment (if you follow). Should
there be a defense against that?  Such as a segment counter compared to
segment number.

Dick

JvanKatwijk, 2017-08-25:
Yes
to do it really good we should take charsets and broken messages into
account. The latter is easy in the dab-cmdline verion,
just check that the ecoming segment follows the last segment and ignores is
we are out of sync
But, while technically fairly easy to address, it shows the more - almost -
philosophical - issue of whether you want only
complete messages or all messages

The Charset and C++ are not easy to handle since there is no real support
for charsets in C++11. In the Qt-DAB version that is easy to handle.

jan

closed 2018-02-12 by JvanKatwijk