JvanKatwijk/dab-cmdline #15: Switch program

Razmuzh, 2017-10-25:
Is there any possibility to switch between the programs in the ensemble without quitting the program first?

JvanKatwijk, 2017-10-26:
How would you switch to another program within the same ensemble without
something to provide userinteraction (i.e. a gui?)

Razmuzh, 2017-10-26:
Maybe via the already open console or via a port?

JvanKatwijk, 2017-10-26:
There is no one listening to the console. What you suggest is to add a
(probably) small GUI. The functionality in the library would allow it,
but you have to build some form of interaction to the main program.
Anyway, the example programs I provide do not support that

Razmuzh, 2017-10-26:
Ok, I'll have to look at that possibility. The radio I'm building doesn't have any other input than a pot-meter for choosing channels, and one for volume. But I plan to play some 'static radionoise' between channel-switching to mask the switch :-)
[image]

JvanKatwijk, 2017-11-10:
Hi
As an experimental gadget I created an example-5 that basically is the same
as example-2, however, extended with a small keyboard
listener that will react on touching the return key by selecting the "next"
program in the ensemble.

closed 2018-02-12 by JvanKatwijk