JvanKatwijk/dab-cmdline #16: Wrong console text

andimik, 2017-11-05:
When I specify `-S` then the text in the console is still your default value from the Netherlands.

`dab-airspy-2 -C 10D -S 9201 -G 100`

shows

```
going to start program Classic FM
selected Slo 1 Prvi      
```

JvanKatwijk, 2017-11-06:
When i am back i will look at it. Which variant are you using?

andimik, 2017-11-08:
It is example 2

JvanKatwijk, 2017-11-08:
you could give it a try now

andimik, 2017-11-08:
Yes, confirmed. The text is now correct.

> there might be a DAB signal here
> 
> we try to start program Slo 3 ARS       
> selected Slo 3 ARS       
> 	startaddress	= 288
> 	length		= 96
> 	subChId		= 3
> 	protection	= 2
> 	bitrate		= 128
> 

But it would also be useful to comment out in `motdata.cpp`, lines 240 and 241 as on the console there is no slideshow working ... ;-)

> going to show picture

And: Could you filter DLs which is repeated, so that only one new DL line is being displayed (instead of continuously writing them)?

```
we have now DAB+
www.rtvslo.si/ars 
www.rtvslo.si/ars 
www.rtvslo.si/ars 
Radio Slovenija
Radio Slovenija
.
.
.
Radio Slovenija
Radio Slovenija
www.rtvslo.si/ars 
www.rtvslo.si/ars 

```

proposal:
```
we have now DAB+
www.rtvslo.si/ars 
Radio Slovenija
www.rtvslo.si/ars 

```

JvanKatwijk, 2017-11-08:
your wish is my command 😁, i.e. I'll look into it forthcoming days

andimik, 2017-11-08:
And: What kind of format is written the `-O` option?

Seems this is not wave, not aac, not eti.

JvanKatwijk, 2017-11-08:
It is an unformatted stream of pcm samples (rate 48000). I vaguely remember
that - at some point in time - I needed it. Using a "-" as filename causes
the data to be sent to stdout

andimik, 2017-11-08:
[image]
Ok, I found it out.

signed 16bit PCM Little Endian
2 channels
48 kHz

Maybe you add that information into the `README` file

JvanKatwijk, 2017-11-09:
seems to be addressed now.

closed 2017-11-10 by JvanKatwijk