JvanKatwijk/dab-cmdline #17: Compiling Simple DAB

andimik, 2017-11-08:
I tried to compile under Ubuntu with Qt-creator, but after 35 seconds it stops compiling:

```
usr/bin/ld: **/tmp/ccM3phWs.ltrans5.ltrans.o**: undefined reference to symbol 'dlclose@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libdl.so.2: error adding symbols: DSO aus der Kommandozeile fehlt
Makefile:276: die Regel für Ziel „linux-bin/simpleDab“ scheiterte
collect2: error: ld returned 1 exit status
make: *** [linux-bin/simpleDab] Fehler 1
20:57:02: Der Prozess "/usr/bin/make" wurde mit dem Rückgabewert 2 beendet.
Fehler beim Erstellen/Deployment des Projekts simple-dab (Kit: Desktop Qt 5.8.0 GCC 64bit)
Bei der Ausführung von Schritt "Make"
20:57:02: Verstrichene Zeit: 00:35.
```

or on console with `make`

```
/tmp/cchtDiBr.ltrans0.ltrans.o: In Funktion `simpleDab::simpleDab(QSettings*, QWidget*) [clone .constprop.38]':
<artificial>:(.text+0x2226): Nicht definierter Verweis auf `dabInit'
/tmp/cchtDiBr.ltrans1.ltrans.o: In Funktion `simpleDab::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)':
<artificial>:(.text+0x9b): Nicht definierter Verweis auf `dabStop'
<artificial>:(.text+0x5c3): Nicht definierter Verweis auf `dabStartProcessing'
<artificial>:(.text+0xb5f): Nicht definierter Verweis auf `dabService'
<artificial>:(.text+0xc1b): Nicht definierter Verweis auf `dabReset'
collect2: error: ld returned 1 exit status
Makefile:186: die Regel für Ziel „linux-bin/simpleDab“ scheiterte
make: *** [linux-bin/simpleDab] Fehler 1

```

there are no files in `/tmp/` starting with cc*.*

----

If I comment out the spectrum,

`#CONFIG		+= spectrum`

the error is

```
simple-dab.cpp:119: Fehler: ‘spectrumHandler’ was not declared in this scope
  spectrumHandler = NULL;
  ^
```

JvanKatwijk, 2017-11-09:
I am not sure what happens here. Did you install the dab library?

andimik, 2017-11-09:
I see. No, just dab-cmdline for wave, airspy and rtl_sdr. 

Will test it in the next days.

andimik, 2017-11-12:
I'll give up.

Have installed the library with

```
cd dab-cmdline/
cd library/
mkdir build
cd build/
cmake .. -DDABSTICK=ON -DRTLTCP=ON -DSPECTRUM=ON -DAIRSPY=ON
make
sudo make install
```

then I went to the simple-dab folder, but added `QMAKE_CXXFLAGS	+= -std=c++11` to `simple-dab.pro` and then tried to build it

```
cd ..
cd simple-dab/
mkdir build
cd build/
qmake ..
make
```

It failed again with 

```
/usr/bin/ld: linux-bin/simpleDab: hidden symbol `_ZN13deviceHandler12set_autogainEb' in /tmp/ccd8DNeV.ltrans2.ltrans.o is referenced by DSO
/usr/bin/ld: final link failed: Ungültiger Wert
collect2: error: ld returned 1 exit status
Makefile:185: die Regel für Ziel „linux-bin/simpleDab“ scheiterte
make: *** [linux-bin/simpleDab] Fehler 1
```

Even a `sudo ldconfig` does not help me.

This is Ubuntu 16.04 LTS

JvanKatwijk, 2017-11-12:
It seems an issue with Ubuntu, on Fedora there is not a problem.
Later forthcoming week I'll try to install Ubuntu in a virtual machine,
then I can run some tests there

best
jan

andimik, 2017-12-18:
I tried to build it again. 

Now it works, but the message in simpleDAB is now `Sorry, no device found`, although I've added this in the library:

`~/dab-cmdline/library/build$ cmake .. -DDABSTICK=ON -DRTLTCP=ON -DSPECTRUM=ON -DAIRSPY=ON -DRAWFILES=ON`

```
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - doneCK
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.4
-- Found FFTW3F: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found zlib: 
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    AIRSPY
    DABSTICK
    RAWFILES
    RTLTCP
    SPECTRUM

-- Build files have been written to: /home/andreas/dab-cmdline/library/build

```

See the sentence: 
> were not used

Does the library accept rawfiles or wavefiles?

JvanKatwijk, 2017-12-19:
Hi Andreas

I think there is a misunderstanding. The basic idea of the example programs
is give people an idea
of what can be done. They were not meant to be full working programs, just
proof of concept.
Given that thre is a library, and looking at the examples one should be
able to create something
to one's own ideas

The simpledab version is no exception to this. It is a demonstration that
the library can be used
with a GUI generating mechanism

The library itself is free from devices, the actual device should be
handled by the demonstration program,
as is the case in example-1 .. example-6. The python example is an
exception to this, since I have no idea
how to connect a device to the python layer, I created an interface layer
between the python part and the
library part, including the device.

Now your question: since simpleDab is an example program, the Qt layer
binds to the sdrplay device.
If you really want to use it with e.g. dabstick or airspy I'll be happy to
create - in source form - the configuration
(not today, since I have to look after my 3 year old granddaughter the
whole day, nice, funny and extremely tiring)

best
jan

andimik, 2017-12-19:
I see. I thought this is a very rudimentary version of qt-dab for laptops with less power.

closed 2017-12-19 by andimik

JvanKatwijk, 2017-12-19:
I added the possibility in configuring  either dabstick, sdrplay or airspy
in the simple-dab.pro file. If you are in need of having wavfiles or
rawfiles added, let me know
it is a matter of minutes (actually tens of minutes) adding these.

best
jan

JvanKatwijk, 2017-12-19:
ps

the load of library based versions is less, buit as far as I have seen,
only marginally