JvanKatwijk/dab-cmdline #19: How to use raw files

athoik, 2017-12-17:
Hi @JvanKatwijk, 

First of all thanks for the great job you are doing with DAB/DAB+.

I would like to know how can I use the samples with raw files (eg like the ones available here: https://sdr.kt.agh.edu.pl/sdrdab-decoder/download.html#download-data)

I would like extend a small plugin for Enigma2 (https://github.com/satdreamgr/SDGRadio) with DAB/DAB+ and your repository seems the most promising.

Currently I have manage to compile example3 (with -DWAVFILES=ON), but samples are not recognized.

```# dab-files-3 -F ./antena-1_dab_229072kHz_fs2048kHz_gain42_1.raw
dab_cmdline V 1.0alfa,
                          Copyright 2017 J van Katwijk, Lazy Chair Computing
opt = F
file ./antena-1_dab_229072kHz_fs2048kHz_gain42_1.raw no legitimate sound file
allocating device failed (24), fatal
```

Once again thank you.

JvanKatwijk, 2017-12-17:
> sdrdab-decoder/download.html#download-data)
> satdreamgr/SDGRadio) with DAB/DAB+ and your repository seems the most

Good question,

The "wavfiles" is for regular sndfiles (PCM files with a rate of 2048000).
Never got the question to add iq or raw files, you are the first. I'll
handle it coming week

best
jan

JvanKatwijk, 2017-12-17:
You could give it a try now. it is called RAWFILES and it compiles well for
example-2, 3, 5

I never use the stdout output, so it might be - unlikely but you never know
- that somewhere some output is written to stdout
(the funny thing is that when running the python example, stderr is hidden,
so I played somewhat with stdout, I think it was
all reset though)

best
jan

athoik, 2017-12-17:
Wow that was fast :)

```
# dab-raw-3 -F ./antena-1_dab_229072kHz_fs2048kHz_gain42_1_long.raw -P "PR Jedynka" > long.stream
dab_cmdline V 1.0alfa,
                          Copyright 2017 J van Katwijk, Lazy Chair Computing
opt = F
opt = P
ofdm word gestart
Period = 8000
PR Tr�jka        (3233) is part of the ensemble
PR Dw�jka        (3222) is part of the ensemble
PR Czw�rka       (3223) is part of the ensemble
Radio Poland     (332A) is part of the ensemble
PR Jedynka       (3211) is part of the ensemble
Radio Rytm       (32A7) is part of the ensemble
Polskie Radio 24 (32A6) is part of the ensemble
ensemble Polskie Radio    is (3203) recognized
Radio Krak�w     (37FC) is part of the ensemble
there might be a DAB signal here

going to start program PR Jedynka
selected PR Jedynka
        startaddress    = 0
        length          = 84
        subChId         = 1
        protection      = 2
        bitrate         = 112
we have now DAB+
Sygnaly Dnia
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
End of file, restarting
Sygnaly Dnia
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.lskie Radio S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
Sygnaly Dnia  S.A.
End of file, restarting
Sygnaly Dnia
Sygnaly Dnia  S.A.
^Clskie Radio S.A.

# cat long.stream  | gst-launch-1.0 fdsrc ! audio/x-raw, format=S16LE, channels=2, layout=interleaved, rate=48000 ! dvbaudiosink
Setting pipeline to PAUSED ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Got EOS from element "pipeline0".
Execution ended after 0:02:46.534099714
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

And it works!

Great job, now I have to see the other examples (scanning, finding channels, etc).

JvanKatwijk, 2017-12-17:
I had nothing serious at hand, and it seemed a rather illogical omission,
so ....

Anyway, have fun

best
jan

closed 2017-12-17 by athoik