JvanKatwijk/dab-cmdline #24: library: install headers

athoik, 2017-12-19:
Hi,

The make install of the library, installs only libdab_lib.so

I think it should install also the headers in /usr/local/include by default.

JvanKatwijk, 2017-12-19:
I have to think about it. It fits in the basic idea of the library,
although comments I got so far all had to do with the example programs.

athoik, 2017-12-19:
My idea was to be able to create something that depends on the library (and required headers to compile it).

Currently the only way is to manually add all sources into a new project that depends on dab-cmdline library.

Having a DAB library is really something nice in order to create standalone applications.

Thank you,
Athanasios

JvanKatwijk, 2017-12-19:
Yes,

but you can obviously copy the include files to your new projects

closed 2018-02-12 by JvanKatwijk