JvanKatwijk/dab-cmdline #32: Handle MOT data

athoik, 2018-01-10:
Hi,

Is there a way on `dab-cmdline` to handle the MOT data.

If not it would be really nice if we can add a "dump folder" where MOT data stored there. Also having a callback in order to inform program about the availability of a new object in dump directory.

Thanks,
Athanasios

JvanKatwijk, 2018-01-10:
You mean the MOT data (i.e. pictures) in the PAD or the more general MOT
data in data services.
The first one might be doable with limited resources

athoik, 2018-01-10:
Hi,

Yes pictures was exactly what I was thinking about.

JvanKatwijk, 2018-01-10:
The issue with pictures is that you need a mechanism to show them , command
line implementations usually do not have such a mechanism.
It is not something that I can do in a lost hour or so, the API will change
one way or another, the interface layer for the Python version has to
change as well.

athoik, 2018-01-10:
Once the picture is available and there is a callback (so I can print something to stderr, like json), then I guess I can handle it just fine!

The output or dab-cmdline (or it this case the sdgradio https://github.com/satdreamgr/dab-cmdline/tree/sdgradio/sdgradio) will be parsed from Enigma2 and the "real" display will be done by the Enigma2 (https://github.com/openpli/enigma2).

Later on, dab-cmdline will be integrated on E2 (hopefully when somebody implements a viterbi  algortithm using NEON instructions, https://github.com/theori-io/nrsc5/blob/master/src/conv_neon.h,  https://github.com/srsLTE/srsLTE/blob/master/lib/src/phy/fec/viterbi37_neon.c).

It's like you are saying, "a never ending work in progress" 👍

JvanKatwijk, 2018-01-11:
> phy/fec/viterbi37_neon.c).

Looking into the issue of making MOT pictures available, I think the best
approach is to
have the picture data written into a (/tmp) file and a signal sent through
a callback function. The signal then indicates that the
picture is written and has has parameter  the name (pathname) of the
picture.
To avoid writing pictures when no one is interested:
a. a configuration parameter, in- or excluding the above mentioned
functionality with conditional compilation, or
b. a command line parameter, passed on to the mot handler (through some
intermediate functions) - that - when set - enables the functionality

My preference would be option (b), although it requires some additional
work.

athoik, 2018-01-11:
I think option (b) is better too. This will make available the feature on demand.

Although conditional compilation wont harm, is somebody needs it, will enable it.

Thanks for your great work 🥇

JvanKatwijk, 2018-01-12:
Something to play with during the weekend

It is slightly different and "alpha" state

The basic idea is that - if a callback is specified - for each slide that
is encoded in the pad, a file is created.
The filename is artificial (some files are unnamed, others are not)

If the callback is absent, no files are written, if the callback is
present, it is called each time
with three parameters: the name (pathname) of the file, the subtype, and
the context (the latter usually being NULL)

So, no switches, no conditional compilation

The spec of the callback is in dab-api.h, in the various examples I
annotated the NULL in either the dabInit or the call to instantiate the
dabclass

athoik, 2018-01-14:
it works great!

Some people from Belgium mentioned that they are receiving slideshow in PAD or in X-PAD.

Is it possible to extract images from there as well?

Thanks!

JvanKatwijk, 2018-01-14:
The pictures you get are from PAD or XPAD

athoik, 2018-01-14:
It might be possible that AAC_OUT is used and no PAD data are handled.

I think this will fix it: https://github.com/satdreamgr/dab-cmdline/commit/9fed1a969d8593db1f565b1f62780f0decb6aee3 but I didn't get any feedback yet.

JvanKatwijk, 2018-01-14:
> I think this will fix it: satdreamgr/dab-cmdline@9fed1a9

I changed the code in mp4 handling in a way that pad handling is decoupled
from the audio handling (did it slightly different than your
suggestion)

pr2git, 2018-01-17:
Hi,

On Belgian DAB/DAB+ we didn't get any "picture".
I have created a capture of the Belgian 12B 225648000 Hz block so I hope that you can find which format is used in Belgium to display slideshows.
From dat2eti:
https://mega.nz/#!iBYjmS7Q!5ZHoCwApNSLZvxtha-Ro2Q_qbMuLjiutbkAmL5sHvp4

Save from: qt-dab-1.0-alpha
https://mega.nz/#!mMQyQSYL!pCVvGd3FK80Is8MZ0vkvXpriEd4Gt4oocq78eraEhlM

Thanks,

Pr2

andimik, 2018-01-17:
Belgium will switch to DAB+ this year, see https://www.worlddab.org/country-information/belgium

Currently they are using a separate stream for the slideshow. See welle.io or ukwtv.de forum. 

Or simply read that: https://github.com/AlbrechtL/welle.io/issues/171

JvanKatwijk, 2018-01-17:
> country-information/belgium
> Or simply read that: AlbrechtL/welle.io#171

A simple check with running the example data in Qt-DAB shows that most
stations are DAB, some are DAB+ (test music and TARMAC) but none carriers
MOT data.
Qt-DAB will handle MOT data in DAB as well as in DAB+ mode, however, I am
not sure that dab-cmdline will handle MOT in DAB  as well
I read that since the end of '17 some programs are transmitted in DAB+, so
may be in the near future ...

pr2git, 2018-01-17:
Hi,
As andimik mentionned and as reported in the different link he provides, belgium is using subchannels to send the Slideshow.
I come to you with this issue because I am using it on a Enigma2 receiver, so I use SDGradio / Enigma2 Radio of athoik.
That's why it would be really nice if you can also extract the slideshow for the belgian channels like you do today with MOT.
I don't think that belgian will change its way to send slideshow (using MOT) since this is also conform to the standard and they broadcast since years now in this way, I have a Medion DAB/DAB+ radio with color LCD and it has no problem with the Belgian channels.  So yes I am 100% sure that there are slideshows today on the DAB+ belgian channels.
Regards,
Pr2

JvanKatwijk, 2018-01-17:
> No, you are wrong. There are no slideshows on the DAB+ channels. Jan
> confirmed it above.
>
> Please do not mix up DAB and DAB+.

There are data channels though that are currently being kept invisible (the
subchannels for the services).
While it is certainly possible making them visible, I did not intend to
make the dab-library handle datra

JvanKatwijk, 2018-01-18:
Yes you are absolutely right,  the subchannels carry MOT data.
One of the advantages of open software is that you have access to the
sources, and you can contribute to the project, not just by asking to have
features added, but
by adding/improving code. On short term I do not have time/opportunity to
add sub channel data handling to the dab-cmdline library, although improved
data handling is
part of the development in the Qt-DAB project

andimik, 2018-01-18:
Jan is right, I cannot write any line of code (never learned it in school), but I try to contribute as much as I can. 

That's Open Source :-)

pr2git, 2018-01-18:
Hi,
I know what OpenSource is and mean and I am also a contributor to other OpenSource projects when I can.
I don't want to hurt anybody, I just provide samples to be analysed to understand what Belgian DAB+ radios are using to broadcast the slideshow.
I was just hoping that will a small change in code it will be possible to extract them.

closed 2018-02-12 by JvanKatwijk