JvanKatwijk/dab-cmdline #43: Program selection selects the wrong program when presented with two very close options

benjojo, 2018-05-13:
```
538              (83C7) is part of the ensemble
100% NL          (83D3) is part of the ensemble
Qmusic           (83C8) is part of the ensemble
BNR Nieuwsradio  (83D1) is part of the ensemble
SLAM!            (82A5) is part of the ensemble
Sky Radio        (83C6) is part of the ensemble
Veronica         (83E1) is part of the ensemble
Radio Maria      (8816) is part of the ensemble
538TOP40         (8802) is part of the ensemble
Radio 10         (83D2) is part of the ensemble
Sublime          (82A8) is part of the ensemble
TPEG 538         (data) (E3880101) is part of the ensemble
Sky Radio Hits   (8808) is part of the ensemble
Sublime You      (8814) is part of the ensemble
Qmusic non-stop  (8804) is part of the ensemble
Radio 10 60s&70s (8818) is part of the ensemble
TPEG 538         (data) (E3880101) is part of the ensemble
GrootNieuwsRadio (8961) is part of the ensemble
```

` ./dab-rtlsdr-4 -C 11C -G 90 -P '538`

selects `538TOP40` rather than `538`

JvanKatwijk, 2018-05-13:
Based on requests the selection of a service was based on a match with the
prefix of the name, which works
in 95% of the cases perfect. What needs to be added is that a full match
has precedence over a match
with a prefix. I'll look into it.

JvanKatwijk, 2018-05-13:
Addressed

I changed the code (actually ony a few lines) such that a "full" match has
precedence over a match with a prefix of the servicename

Of course, if given a prefix of a service name that matches with more than
one service, the choice of the selected service remains a magic one.

2018-05-13 9:26 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> Based on requests the selection of a service was based on a match with the
> prefix of the name, which works
> in 95% of the cases perfect. What needs to be added is that a full match
> has precedence over a match
> with a prefix. I'll look into it.
>
>
>
>> 538              (83C7) is part of the ensemble
>> 100% NL          (83D3) is part of the ensemble
>> Qmusic           (83C8) is part of the ensemble
>> BNR Nieuwsradio  (83D1) is part of the ensemble
>> SLAM!            (82A5) is part of the ensemble
>> Sky Radio        (83C6) is part of the ensemble
>> Veronica         (83E1) is part of the ensemble
>> Radio Maria      (8816) is part of the ensemble
>> 538TOP40         (8802) is part of the ensemble
>> Radio 10         (83D2) is part of the ensemble
>> Sublime          (82A8) is part of the ensemble
>> TPEG 538         (data) (E3880101) is part of the ensemble
>> Sky Radio Hits   (8808) is part of the ensemble
>> Sublime You      (8814) is part of the ensemble
>> Qmusic non-stop  (8804) is part of the ensemble
>> Radio 10 60s&70s (8818) is part of the ensemble
>> TPEG 538         (data) (E3880101) is part of the ensemble
>> GrootNieuwsRadio (8961) is part of the ensemble
>>
>> ./dab-rtlsdr-4 -C 11C -G 90 -P '538
>>
>> selects 538TOP40 rather than 538
>>
>> <https://github.com/JvanKatwijk/dab-cmdline/issues/43>, or mute the
>> thread
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980
> +31 (0) 628260355

benjojo, 2018-05-14:
Cool, Fixed!

closed 2018-05-14 by benjojo

fsfhiyg, 2019-02-01:
May 2

The return of a crypto.Signer.Public() must be a pointer to a key

Otherwise various libraries can't use it. I don't know if this library supports DSA keys or something, but I could not find any evidence of that, s…

+12 −5  •3 comments

JvanKatwijk, 2019-02-01:
can you give an example