JvanKatwijk/dab-cmdline #44: Example-3 no longer works

bartpeeters, 2018-07-22:
First of all thanks for this great repository, I have a SDRPLAY and have been using it to play DAB+ using this repo.

I have been testing using code based on the example-3. I see that you've been busy upgrading the library to a newer version. This seems to have broken example-3.

The CMakeLists.txt of example-3 has a reference to a file that is renamed:
https://github.com/JvanKatwijk/dab-cmdline/blob/master/example-3/CMakeLists.txt#L336

I changed this to `../library/src/backend/data/mot/mot-handler.cpp` and it compiles again. When I run the example it gets to the point where it dab_service(a radio station), but there is no data written to stdout and sometimes I also get a segmentation fault. 

Any help would be appreciated.

JvanKatwijk, 2018-07-22:
> master/example-3/CMakeLists.txt#L336

Yes, the CMakefile contained an error,  the upgrade was twofold, the
mothandling is brought inline with that of qt-dab and the handler for the
sdrplay
is modified since library 2.13 is slightly incompatible with the previous
versions.

I just ran a quick test - right now I am somewhat limited since I am not at
home - but I do not have a problem.

It would be helpful if you could run uding gdb and if there is a
segmentation fault show thestacktrace

Since tonight I am hoe again, I will have a closer look tomorrow

bartpeeters, 2018-07-22:
```
Thread 1 "dab-rtlsdr-3" received signal SIGSEGV, Segmentation fault.
0x00007ffff64b9fff in _int_malloc () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff64b9fff in _int_malloc () from /usr/lib/libc.so.6
#1  0x00007ffff64bb67a in malloc () from /usr/lib/libc.so.6
#2  0x00007ffff6e31159 in operator new (sz=1) at /build/gcc/src/gcc/libstdc++-v3/libsupc++/new_op.cc:50
#3  0x00005555555792f6 in dataBackend::dataBackend(packetdata*, void (*)(unsigned char*, short, unsigned char, void*), void (*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, void*), void*) ()
#4  0x0000555555579792 in mscHandler::set_dataChannel(packetdata*) ()
#5  0x000055555556d56b in dabClass::dab_service(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()
#6  0x0000555555564571 in main ()
```

That's the segfault I get. Sometimes it does run and outputs, sometimes it runs but doesn't output and other times I get the segfault.

JvanKatwijk, 2018-07-23:
Thanks, it turned out that there was an initialization error somewhere. The error showed up with
DAB rather than DAB+ programs. This one seems fixed

closed 2018-07-23 by JvanKatwijk