JvanKatwijk/dab-cmdline #47: mp2processor class does not initialize ctx member variable

borine, 2018-08-06:
When using the "UserData" field in callbacks, I get a segfault playing DAB audio. I found that this is because the userdata pointer i am given in the callback is null.

It turns out this is because the constructor in /library/src/backend/audio/mp2processor.cpp does not initialize the ctx member variable of the mp2processor object. By adding
this->ctx = ctx;
to the constructor body the segfault goes away and I get really faultless audio from the BBC national ensemble in the UK using an RPi 2B+ and a cheap RTL SDR USB device.

borine, 2018-08-12:
The fix for this has been lost in the recent changes for example 10. Could you possibly re-apply it?
Thanks.

JvanKatwijk, 2018-08-12:
done

borine, 2018-08-16:
Thanks - all working again now

closed 2018-08-16 by borine