JvanKatwijk/dab-cmdline #52: is there a way to use this with SDR#

veso266, 2018-12-11:
Hi was wondering if you have any idea how could I use this with SDR# (I was thinking of writing a plugin to output I/Q data out), but then don't know how would I pipe that into dab-cmdline

Thanks for Anwsering and Best Regards

JvanKatwijk, 2018-12-11:
Interesting thought. If you mean "replace the input device by an external
agent" then, sure it is possible.
I do not know SDR#, so I do not know what the possibilities/impossibilities
are, bus, as an example, the software
supports the use of a dabstick over a tcp connection.
The interface in dab-cmdline to the devices is simple, just a handful
commands (set frequency, set gain etc)
and a few requests (i.e. is there data available and get me some data)

veso266, 2018-12-11:
I was thinking of replacing the input with I/Q data to STDin, don't know if I can do that

JvanKatwijk, 2018-12-11:
the question is: who is in control. It is as such fairly easy to make stdin
(or whatever channel) as input and feed that into the "dab-engine",
The "main  program" now does a series of steps. Now suppose you have the
input from stdin, then there is still need for a program to
start up the "dab-engine", to wait for time and frequency synchronization
and for extracting the relevant data of the selected service
from one part, and instructing another part to process the input according
to this data description.

If you would be satisfied with a set up as currently in e.g. example 2,
i.e. a  main program  with command line settings
for the selection of the service (but of course not for device specific
settings, e.g. frequency, gain) with input from stdin rather than some
device
that is certainly possible.

veso266, 2018-12-11:
Thanks will dig into example 2

JvanKatwijk, 2018-12-11:
It is just writing a simple driver that reads in from stdin, you might have
a look at the driver for raw files. Next week I ll have time to write
something like that,

JvanKatwijk, 2018-12-12:
I had some spare time this morning and created an "example 7" in the
dab-cmdline tree that just takes
its input from stdin.
In the conversion I assume that the samples are 8 bit samples, format a la
dabstick, so each sample is a pair of 8 bit values, each in
the range 0 .. 255

Hope you can move on then

veso266, 2018-12-22:
I gave example-7 a go, but first I tried to compile it with mingw32 (its included in msys2)
after fighting with libraries and writing instructions along the way
here they are

1. Download Precompiled FFTW: http://fftw.org/install/windows.html (ftp://ftp.fftw.org/pub/fftw/fftw-3.3.5-dll32.zip)
	copy .h files to mingw32/include
	rename libfftw-3 to libfftw
	rename *.dll to *.dll.a
	Copy *.dll.a files to msys64/mingw32/lib 
Compiling FFTW with MinGW32

	wget ftp://ftp.fftw.org/pub/fftw/fftw-3.3.8.tar.gz
	tar -xvz fftw-3.3.8.tar.gz
	cd fftw-3.3.8
	./configure
	make -j5 (to use all CPU cores it compiles faster)

2. Build libfaad

get libfaad2.dll somewhere?? (dream install dir)
rename libfaad2.dll to libfaad2.dll.a and copy to msys64/mingw32/lib
copy faad.h and neaacdec.h to mingw32/include

3. Build libsamplerate
	git clone https://github.com/erikd/libsamplerate.git
	cd libsamplerate
	./configure
	make -j5 (to use all CPU cores it compiles faster)

	copy C:\msys64\tobuild\DAB\dab-cmdline-master\libs\libsamplerate\libsamplerate-master\src\.libs to mingw32/lib
	copy C:\msys64\tobuild\DAB\dab-cmdline-master\libs\libsamplerate\libsamplerate-master\src\*.h to mingw32/include

4. Building portAudio
	read here: https://app.assembla.com/wiki/show/portaudio/Notes_about_building_PortAudio_with_MinGW
	wget http://www.portaudio.com/archives/pa_stable_v190600_20161030.tgz
	tar -xvz pa_stable_v190600_20161030.tgz
	cd portaudio
	./configure
	make -j5 (to use all CPU cores it compiles faster)
	cd lib/.libs
	copy libportaudio.dll.a to mingw32/lib
	copy include/*.* to mingw32/include

3. Building example-7
	git clone https://github.com/JvanKatwijk/dab-cmdline.git
	cd dab-cmdline/example-7
	mkdir build
	cd build
	cmake ../ -G "MSYS Makefiles"

I finaly went to compiling example-7
not sure how are you compiling that, did you tried doing it on windows?
but I got a lot of errors :( about conflicting declaration of time_t, but it only happened to people on linux cross-compiling for windows https://bbs.archlinux.org/viewtopic.php?id=197775

not sure if you have any idea
```bash
[  2%] Building CXX object CMakeFiles/dab_cmdline-7.dir/C_/msys64/tobuild/DAB/dab-cmdline-master/libs/dab-cmdline/devices/stdin-handler/stdin-handler.cpp.obj
In file included from C:/msys64/usr/include/stdio.h:61:0,
                 from C:/msys64/tobuild/DAB/dab-cmdline-master/libs/dab-cmdline/devices/stdin-handler/stdin-handler.cpp:25:
C:/msys64/usr/include/sys/types.h:129:18: error: conflicting declaration 'typedef __int_least64_t time_t'
 typedef _TIME_T_ time_t;
                  ^~~~~~
In file included from C:/msys64/mingw32/i686-w64-mingw32/include/stddef.h:7:0,
                 from C:/msys64/mingw32/lib/gcc/i686-w64-mingw32/7.3.0/include/stddef.h:1,
                 from C:/msys64/usr/include/sys/cdefs.h:45,
                 from C:/msys64/usr/include/stdio.h:35,
                 from C:/msys64/tobuild/DAB/dab-cmdline-master/libs/dab-cmdline/devices/stdin-handler/stdin-handler.cpp:25:
C:/msys64/mingw32/i686-w64-mingw32/include/crtdefs.h:136:20: note: previous declaration as 'typedef __time32_t time_t'
 typedef __time32_t time_t;
                    ^~~~~~
[... 129 lines cut ...]
C:/msys64/tobuild/DAB/dab-cmdline-master/libs/dab-cmdline/devices/stdin-handler/stdin-handler.cpp:86:5: note: suggested alternative: '_sleep'
     usleep (100);
     ^~~~~~
     _sleep
C:/msys64/tobuild/DAB/dab-cmdline-master/libs/dab-cmdline/devices/stdin-handler/stdin-handler.cpp: In member function 'virtual void stdinHandler::run()':
C:/msys64/tobuild/DAB/dab-cmdline-master/libs/dab-cmdline/devices/stdin-handler/stdin-handler.cpp:113:8: error: 'usleep' was not declared in this scope
        usleep (100);
        ^~~~~~
C:/msys64/tobuild/DAB/dab-cmdline-master/libs/dab-cmdline/devices/stdin-handler/stdin-handler.cpp:113:8: note: suggested alternative: '_sleep'
        usleep (100);
        ^~~~~~
        _sleep
C:/msys64/tobuild/DAB/dab-cmdline-master/libs/dab-cmdline/devices/stdin-handler/stdin-handler.cpp:126:8: error: 'usleep' was not declared in this scope
        usleep (nextStop - getMyTime ());
        ^~~~~~
C:/msys64/tobuild/DAB/dab-cmdline-master/libs/dab-cmdline/devices/stdin-handler/stdin-handler.cpp:126:8: note: suggested alternative: '_sleep'
        usleep (nextStop - getMyTime ());
        ^~~~~~
        _sleep
make[2]: *** [CMakeFiles/dab_cmdline-7.dir/build.make:63: CMakeFiles/dab_cmdline-7.dir/C_/msys64/tobuild/DAB/dab-cmdline-master/libs/dab-cmdline/devices/stdin-handler/stdin-handler.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/dab_cmdline-7.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```
it looks like c++ compiling is interfering with standard mingw include files or something like that

JvanKatwijk, 2018-12-23:
I never compiled it under Windows, I am afraid.
The "stdin-handler" was written with Linux in mind.
Some of the errors here are easy to circumvent. The approach is pretty
simple:
start a separate thread, reading in chunks of input, each chunk -  which is
read-in synchronously, that is why the separate thread
is needed - consisting of 2048 I/Q samples (i.e. one millisecond of data.

The time functions are there to delay in case the data is too fast. If you
*know*  that - since you are using some other program
to collect the data, that the samplerate is just 2048000, then these time
functions can be eliminated of course.

That leaves the problem of the thread.
I have taken the - more or less - standard C++11 approach to threading, it
is, however, easy to change that to a standard POSIX model

The errors that sound more serious to me are the ones related to the
complex data type, that is spread all over most sources since
it is standard C++

veso266, 2018-12-23:
Thanks for great explanation, now I just need to fix it :smile:

closed 2019-02-01 by JvanKatwijk