JvanKatwijk/dab-cmdline #59: Stuttering audio

maartenlambrecht, 2019-04-02:
I'm having some trouble with this dab-cmdline application on Raspbian (RPi 3b+).
When listening to a DAB+ radio station in Belgium, I always get **stuttering audio**.
When using Dablin together with eti-cmdline I get also stuttering audio.
On Windows I get nice sound without any hickups. So it isn't a reception issue.

What could be causing this problem?

Used command:
`./dab-rtlsdr-3 -C 12A -P Klara -Q | aplay -r 48000 -f S16_LE -t raw -c 2 -v`

Output:
```dab_cmdline V 1.0alfa,
                          Copyright 2017 J van Katwijk, Lazy Chair Computing
opt = C
opt = P
opt = Q
going for rtlsdr 223936000 45
OK, functions seem to be loaded
Found Fitipower FC0012 tuner
samplerate set to 2048000
Supported gain values (5): -9.-9 -4.0 7.1 17.9 19.2 
effective gain: gain 7.1
ensemble DAB+ VRT         is (6001) recognized
Radio 1          (6301) is part of the ensemble
Klara            (6303) is part of the ensemble
Radio 2 LIM      (6602) is part of the ensemble
Studio Brussel   (6304) is part of the ensemble
Radio 2 O-VL     (6C02) is part of the ensemble
Radio 2 ANT      (6502) is part of the ensemble
TPEG             (data) (E0600001) is part of the ensemble
Radio 2 W-VL     (6B02) is part of the ensemble
Radio 2 VL-BR    (6402) is part of the ensemble
MNM              (6305) is part of the ensemble
VRT NWS          (631E) is part of the ensemble
MNM Hits         (631F) is part of the ensemble
Klara Continuo   (631A) is part of the ensemble
there might be a DAB signal here

we try to start program Klara
protection handler is eep_protection
we have now DAB+
Playing raw data 'stdin' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Plug PCM: Hardware PCM card 0 'bcm2835 ALSA' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 24000
  period_size  : 6000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 6000
  period_event : 0
  start_threshold  : 24000
  stop_threshold   : 24000
  silence_threshold: 0
  silence_size : 0
  boundary     : 1572864000
  appl_ptr     : 0
  hw_ptr       : 0`

JvanKatwijk, 2019-04-03:
Thanks for the message. Did you have a look at the processorload? I
(vagely) remember that in the combination you are using
the load on one of the cores was too high. There are three possible causes
a. the signal is weak and not all frames can be recognized and/or decoded
b. there is an overload on one (or more) of the processor cores
c. the audio subsystem needs larger buffers - due to irregular arrival of
data

I'll run the combination later today (I usually use the built in portaudio
driver)

JvanKatwijk, 2019-04-03:
I ran some tests this morning and I found that the example-3 + aplay gives
occasionally a "hickup" in the sound output,
(i.e. one or twice a minute), the example-2 did not give a single one in 15
minutes listening.
I uncommented the line in the function "mscQuality" in example-3. The
function will give once a second or so, the
quality (three terms to be explained later) and I notices that the
stuttering in example-3 ocurred when the
"fe" output of the mscQuality function is less than 100.

fe indicates the percentage of DAB+ frames identified (a DAB+ frame
consists of 5 consecutive DAB frames),
if the percentage is less than 100 it just means that not all DAB+ frames
could be identiftied
(the rsE and aacE numbers give the sucessrate of the rs error repair and
the - final - aac Decoding

Now in my situation I am using an SDRplay device as "hardware", which is
superior to a DABstick

So, to me it seems a signal quality issue
static
void    mscQuality      (int16_t fe, int16_t rsE, int16_t aacE, void *ctx) {
//      fprintf (stderr, "msc quality = %d %d %d\n", fe, rsE, aacE);
}
best
jan

maartenlambrecht, 2019-04-03:
Thank you for your quick response!
I did some further research:

The CPU usage is indeed pretty high.[ 'top' shows a usage of 103 %CPU](https://botervlieg.be/public/TEMP/dab.png)

Testing Welle.io does work as expected, withouth any hickups. But it takes some time to 'get a fix'.
`welle-cli -c 12A -w 7979 -C 1 -g -1`
 ([before fix](https://botervlieg.be/public/TEMP/pre.png)) ([after fix](https://botervlieg.be/public/TEMP/post.png))

My SNR is 6 to 8. Should be ok i think.

Using Dablin together with eti-cmdline in commandline I get one second good music, then the hickups begin. In the terminal [alot of (FIB)'s start to show.](https://botervlieg.be/public/TEMP/fib.png)

andimik, 2019-04-03:
Just a quick remark

Your screenshots show lots of errors. Don't you have a better signal in Belgium?

JvanKatwijk, 2019-04-04:
I really would suggest to run for some time with the afore mentioned line
in the code uncommented
I ran a test for half an hour (the example-3, with an sdrplay though) and
each "hickup" of the sound
(actually only a very few) could be linked to the software not being able
to identify a dab+ frame, i.e. a disturbed signal

maartenlambrecht, 2019-04-17:
Thank you for your reply.
I will try to optimize my reception with a better antenna, and will then do some more tests.

closed 2019-05-01 by JvanKatwijk