JvanKatwijk/dab-cmdline #60: Which ha?

teicors, 2019-04-10:
Ho, this is note an issue, but I would  know  which  kind of hw  I must use in order  to tune  a dab  station. May  be  I'm wrong,  but I wasn't able  to find any direction in the files.

Regards, 
Angelo

JvanKatwijk, 2019-04-10:
Well
The dab-cmdline software can be used with RT2832 based dabsticks (cheap),
sdrplay devices (more expensive),
airspy devices (also more expensive) and hackrf devices (pretty expensive)

teicors, 2019-04-10:
Hi Jvan,
I've to study better your code.... I plugged my usb stick (rt2832u), but
the "scan" wasn't able to find it. The lsusb show me the stick.

Thanks for the fastest answer.

Angelo

JvanKatwijk, 2019-04-10:
One of the issues with simple devices like the rt2832 based dabsticks is to
have the gain setting right
You provide all parameters through the commandline, while with e.g. qt-dab
you can change the settings
while running
'

closed 2019-05-01 by JvanKatwijk