JvanKatwijk/dab-cmdline #61: cmake include_directories: /usr/include breaks cross-build

borine, 2019-06-05:
Many thanks for developing and sharing this great software. It works well for me, but there is one irritation with the cmake configuration files that means I always have to patch them before I can build.
Is there a specific reason why many of the CMakeLists.txt files in this project explicitly add /usr/include to the "include_directories" ? I ask because this breaks cross-compilation, causing the cross compiler to look in the build system's native /usr/include folder before the desired sysroot /usr/include.
I can't imagine there is any sane c++ toolchain that does not automatically search for system header files in /usr/include, so my preferred fix is to just delete that line from all CMakeLists.txt files.
If there is some obscure reason why dab-cmdline really does need to explicitly tell the compiler to search /usr/include, then can I ask that the line be edited to "=/usr/include" as that would tell the compiler to use the correct sysroot?
I can prepare a pull request with either of the above 2 changes if you are happy to accept either of them.

JvanKatwijk, 2019-06-06:
Good question. I suppose that when writing the cmakelists file I added it
and since it did not harm, there was never a good reason to remove. Right
now I am not at home so I cannot test or change anything. End of the month
I ll address to issue

andimik, 2021-05-20:
@borine 

This issue from 2019 is still open. Was there any progress in the meantime?

borine, 2021-05-20:
I haven't used this project for a while now, but a quick look at the sources shows that CMakeLists.txt in several directories does still have `/usr/include` in the `include_directories` list. So no, there has been no progress. The fix is just to delete that entry from all `include_directories` lists - but I guess no-one else has reported any problems with this, so perhaps its only my build environment that fails.

JvanKatwijk, 2021-05-20:
The README of dab-cmdline clearly states that the examples are just to help
potential users of the dab-library or its sources in using it.
They are not supported

borine, 2021-05-20:
As no-one else has reported any problems with this, and I'm not using DAB at the present time, might as well close this. It is most unusual to tell a compiler where to find its own host system include files though :confused:

closed 2021-05-20 by borine