JvanKatwijk/dab-cmdline #63: Problem with main.cpp building example-3

montgomeryb, 2019-08-09:
Hello, excellent project, thank you. In trying to build example-3, I ran in to a problem:
`
[  4%] Building CXX object CMakeFiles/dab-rtlsdr-3.dir/main.cpp.o
src/dab-cmdline/example-3/main.cpp: In function ‘int main(int, char**)’:
src/dab-cmdline/example-3/main.cpp:324:11: error: ‘theGain’ was not declared in this scope
           theGain = atoi (optarg);
`

After looking at some of the code, I added a definition in to main.cpp:
`
int             deviceGain      = 45;   // scale = 0 .. 100
int             theGain         = deviceGain;
`
This seemed to resolve the issue.
Regards,
Bryan

JvanKatwijk, 2019-08-09:
Hello

Yes I am afraid some of the main programs became a little messy by
incrementally supporting devices, each with different settings for gain.
I myself am using example 2, that is reasonably OK.  I'll give it a try
next week - have to make some other changes as well -
Thanks for the message

best
jan

JvanKatwijk, 2019-08-11:
I rewrote (part of the) main program for example 3, such that the command options are the same
as for example 3 (of course with the exception of the setting of the audio card). Furthermore now
the LimeSDR is part of the configuration

closed 2019-08-11 by JvanKatwijk

montgomeryb, 2019-08-11:
Hello,
I just pulled the latest version. I did have a problem compiling it that
took me a while to find out why .... I did not have the en_us.UTF8 locale
defined so it gave me:
dab_cmdline example III,
                          Copyright 2017 J van Katwijk, Lazy Chair Computing
terminate called after throwing an instance of 'std::runtime_error'
  what():  locale::facet::_S_create_c_locale name not valid
Aborted

After doing some research, I configured the locale and the program runs
fine. Now, I just need to get a consistent DAB signal with my setup.

Thanks for the update. My end goal at the moment is to hope to be able to
stream a DAB channel to icecast, but again I first need to improve my
reception.

Bryan.

JvanKatwijk, 2019-08-12:
Yes, I think the locale lines should be changed to just calling in the
"local" locale std::setlocale (LC_ALL, "");

jan