JvanKatwijk/dab-cmdline #64: programDataHandler does not trigger

athoik, 2020-04-13:
Hi, 

I hope you are fine!

I am testing latest version of dab-cmdline and notice a minor issue. 
The programDataHandler doesn't trigger any more.

As a workaround when selecting a new audio, I am calling programDataHandler(ad, null)  from program.

Best Regards,
Athanasios

JvanKatwijk, 2020-04-13:
Hello
It was a long time since your last message!
But yes, you are right. No idea what happened, but your solution is the
best.
It makes sense to couple it to the selection of the service, then the data
will show.
I have no idea when it was changed or what happened.

All the best
jan

JvanKatwijk, 2020-04-14:
should be fixed now

athoik, 2020-04-14:
Latest commit doesn't fully restore the program data handler. 

I took the liberty to fix missing parts on #65.

closed 2020-04-15 by athoik