JvanKatwijk/dab-cmdline #66: mot-object: does not save object to filesystem

athoik, 2020-04-14:
Hi, 

It seems that mot-object doesn't save images to filesystem any more.

https://github.com/JvanKatwijk/dab-cmdline/blob/252ab7fd2665e1ebd55fa7e887d20df4c1e1b2fa/library/src/backend/data/mot/mot-object.cpp#L163-L173

It was disabled on: https://github.com/JvanKatwijk/dab-cmdline/commit/d858f4f49cb5f04a2dd959959b27cb68f3f4a33c

(but before disable, it was written on working directory, instead of /tmp)

I restore it like this:

https://github.com/satdreamgr/dab-cmdline/blob/dev/library/src/backend/data/mot/mot-object.cpp#L164-L176

Obviously if there is no motdataHandler we can simply return immediately from handleComplete.

Restore this functionality?

JvanKatwijk, 2020-04-15:
> It was disabled on: d858f4f

I do not have a problem with that, the reason the objects were not saved
was a user request to stop automatic saving.
Actually I think it should be done in the user handler, rather than from
deep inside the code
I'll have a good look

JvanKatwijk, 2020-04-15:
I changed the code:
if a handler is present then the object is written as a file in /tmp
if no handler is preset, object is ignored

closed 2020-04-15 by JvanKatwijk

athoik, 2020-04-15:
And that perfectly makes sense! Thank you!