JvanKatwijk/dab-cmdline #67: AAC_OUT: it seems AAC LC format changed to AAC SBR

athoik, 2020-04-14:
Using latest version I am getting the following output on AAC ouput:

```
# ffprobe -i rad.aac
ffprobe version N-52056-ge5d25d1147-static https://johnvansickle.com/ffmpeg/  Copyright (c) 2007-2020 the FFmpeg developers
  built with gcc 6.3.0 (Debian 6.3.0-18+deb9u1) 20170516
  configuration: --enable-gpl --enable-version3 --enable-static --disable-debug --disable-ffplay --disable-indev=sndio --disable-outdev=sndio --cc=gcc --enable-fontconfig --enable-frei0r --enable-gnutls --enable-gmp --enable-libgme --enable-gray --enable-libfribidi --enable-libass --enable-libfreetype --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-librubberband --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libvorbis --enable-libopus --enable-libtheora --enable-libvidstab --enable-libvo-amrwbenc --enable-libvpx --enable-libwebp --enable-libx264 --enable-libxml2 --enable-libxvid --enable-libzimg
  libavutil      56. 42.101 / 56. 42.101
  libavcodec     58. 76.100 / 58. 76.100
  libavformat    58. 42.100 / 58. 42.100
  libavdevice    58.  9.103 / 58.  9.103
  libavfilter     7. 77.100 /  7. 77.100
  libswscale      5.  6.101 /  5.  6.101
  libswresample   3.  6.100 /  3.  6.100
  libpostproc    55.  6.100 / 55.  6.100
[aac_latm @ 0x27a0ae0] SBR with 960 frame length is not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.
[aac_latm @ 0x27a0ae0] SBR with 960 frame length is not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.
Input #0, loas, from 'rad.aac':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Audio: aac_latm (LC), 24000 Hz, stereo, fltp
[aac_latm @ 0x27a1400] SBR with 960 frame length is not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.
```

I suspect that issue occurs since last change on mp4processor.

https://github.com/JvanKatwijk/dab-cmdline/commit/70c978634b6d0ab66500f31a7d90f605f533337b#diff-116d5b7495d6f9297530f92514c37515

How can we verify if DAB+ uses AAC SBR or AAC LC?

Because if ERT DAB+ uses AAC SBR, then it's something missing from ffmpeg :(

andimik, 2020-04-14:
https://www.fmlist.org/sendertabelle/dab-ww_more.php?inputLand=GRC&inputEn=E000

ERT has never used LC as far as I can remember.

Your output tells me that this is definitely wrong. For example 24 kHz instead of 48.

Can you force it to latm instead of loas?

JvanKatwijk, 2020-04-15:
can you run the aac file with VLC?

athoik, 2020-04-15:
Correct! With VLC I have no problem playing the output file produced using AAC_OUT.

On embedded device where project is in use, the GStreamer doesn't recognize it any more, and ffmpeg as well. 

So the only option is to use PCM samples (and those are working great!)

athoik, 2020-04-18:
Hi, 

I record few seconds IQ samples from ERT DAB 12C. [Greece_ERT_12C.iq 103.3 MB](https://mega.nz/file/eIt0QCbZ#WFARymL7e99SoEPmJ9GT9lvTF7sSqqcMXk272LZzlfk)

```
rtl_sdr -f 227360000  -p 23 12C.iq
```

Then used dab-cmdline that outputs AAC (using commit dd3b4660cc38fd9a02a2f1c4fc629f314614ea3a).

```
dab-raw-sdgradio -F 12C.iq -S 1004  1>12C_kosmos_old.aac
ffprobe -i 12C_kosmos_old.aac
...
Input #0, aac, from '12C_kosmos_old.aac':
  Duration: 00:00:12.06, bitrate: 95 kb/s
    Stream #0:0: Audio: aac (HE-AAC), 48000 Hz, stereo, fltp, 95 kb/s
```

Using latest version dab-cmdline:

```
dab-raw-sdgradio -F 12C.iq -S 1004 1>12C_kosmos.aac
Input #0, loas, from '12C_kosmos.aac':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Audio: aac_latm (LC), 24000 Hz, stereo, fltp
[aac_latm @ 0x94050] SBR with 960 frame length is not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.
```

andimik, 2020-04-18:
Is the speed correct in the old aac recording? I guess it is 6.66% slower.

athoik, 2020-04-18:
VLC player handles new format properly. The old format is playing slower. 

So I guess old version didn't properly create AAC.

New version properly creates AAC, but FFmpeg and GStreamer cannot handle it yet.

athoik, 2020-04-18:
FFmpeg although it reports an error, it can transcode sample! 

```
ffmpeg -i 12C_kosmos.aac 12C_kosmos.wav
....
[aac_latm @ 0x7e120] SBR with 960 frame length is not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.
[aac_latm @ 0x7e120] SBR with 960 frame length is not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.
Error while decoding stream #0:0: Invalid data found when processing input
size=    2280kB time=00:00:24.32 bitrate= 768.0kbits/s speed= 107x
video:0kB audio:2280kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.003341%
```

So I guess we need to report that issue to FFmpeg developers.

andimik, 2020-04-18:
It was valid AAC MP4/ISOBMFF container, but it's tricky as DAB+ uses 960 frame length (instead fo 1024). 

For some reasons, the "newer" filetype in LATM/LOAS container format makes more sense.

See https://github.com/Opendigitalradio/dablin , chapter "Recording a service".

Please forget ffmpeg, this is not solved for many years now for HE-AAC, see https://trac.ffmpeg.org/ticket/1407

If you report it, good luck!

athoik, 2020-04-18:
>The AAC recordings can be player e.g. by VLC player. FFmpeg will play them as well, but currently does not support the SBR extension together with the 960 samples per frame transformation and will thus only play the AAC core then.

Ok, so the change is an improvement.

So sad that both GStreamer and FFmpeg cannot handle such format.

closed 2020-04-18 by athoik

aerogus, 2023-07-05:
It seams that a new patch has been proposed in ffmpeg to make compatibility with 960 frame length and SBR (not tested)
https://patchwork.ffmpeg.org/project/ffmpeg/patch/14a406d5-5c56-ef89-bebf-18c205cae59e@walliczek.de/