JvanKatwijk/dab-cmdline #73: Stack corruption in reedSolomon::decode_rs

srcejon, 2021-04-21:
Hi,

It looks like there is stack corruption in reedSolomon::decode_rs().

It has:

uint8_t	omega	  [nroots];

Then calls computeOmega()

Which does:

omega [nroots] = codeLength;

omega [nroots] is beyond the end of the allocated array (which of course only goes to nroots-1).

JvanKatwijk, 2021-04-22:
That is a good one!
I'll have it fixed.

Btw, I was modifying some code in the library (and some of the example
programs), since I was somewhat unhappy with passing
individual functions through the who;e computing chain. Change will be
a (major) the dabInit and other functions take as parameter a struct
API_struct with as fields the various callback functions,
b (minor) I added a callback function that can be used to print the TII
data (major, minor)

On the todo list is a rewrite of example 5 (the one with keyboard "up" and
"down" through the list), where a simple message queui
is used to handle the  result of the callbacks.

srcejon, 2021-04-22:
Great - note that in the MacOS patch you just merged. I'd increased the size of the array by 1 as a temporary workaround.

JvanKatwijk, 2021-04-22:
yes I've seen that!
When I find some time I'll dive into the rs code (has been a long time)

JvanKatwijk, 2021-04-22:
Interesting is that in qt-dab the size was adapted, similar to the size of
lambda, also nroots+1

closed 2021-04-27 by JvanKatwijk