JvanKatwijk/dab-cmdline #77: streamInit() failed in python-example

gvanem, 2021-04-26:
Hello gain. I've been playing around with all the examples (using MSVC and clang-cl).
Now I've patched an successfully linked `libdab_lib.pyd` (using Python 3.6.5).

Building with `HAVE_SDPLAY` and using `py -3 cmdline.py -C12D`, I get a weird error from `mir_sdr_StreamInit()`:
```
sdrdevice found = 1803049E94, hw Version = 255
device installed
Error 3 on streamInit
Error 3 on streamInit
restart failed
no ensemble found, giving up
```

Error `3` is `mir_sdr_OutOfRange`. Which I assume is frequency out-of-range.
First I used `theDevice->defaultFrequency();` -> failed with `Error 3 on streamInit`.
I then patched `startProcessing_p()` to say just:
```c
	b = theDevice	-> restartReader (220);
```
thinking it should be in MHz, but no. The same  `Error 3 on streamInit`.

So how is the frequency supposed to be set?

gvanem, 2021-04-26:
Okay. so the function should something like:
```c
PyObject *startProcessing_p (PyObject *self, PyObject *args) {
PyObject	*handle_capsule;
void		*handle;
PyObject	*result;
bool	b;

	PyArg_ParseTuple (args, "O", &handle_capsule);
	handle = PyCapsule_GetPointer (handle_capsule, "library_object");

	b = theDevice	-> restartReader (MHz(220));
	if (b) {
	   fprintf (stdout, "device restarted\n");
	   dabStartProcessing (handle);
	}
	else
	   fprintf (stdout, "restart failed\n");
	result = Py_BuildValue ("(b)", b);
	Py_INCREF (result);
	return result;
}
```
Seems to work, but still no joy:
```
device installed
device restarted
no ensemble found, giving up
```

Any way of adding some debug-output? If so where and how?

And again, both my RSP1A and RTLSDR stick works fine in Qt-DAB and Welle-IO.

JvanKatwijk, 2021-04-26:
Interesting problem.
As far as I can see, the frequency is computed from the channel name on
initializing the device (dabInit_p, about line 435)
restartReader is called on 2 places, the first one is in the dabInit_p
function (line 488), the second one in the
function startProcessing  (about line 516)

If you make the variable frequency, declared at the start of the dabInit_p
function global, then its value can be used
in both occurrences of restartReader

Years ago that I looked into this code! (actually I was too lazy too delete
it)

closed 2021-04-27 by JvanKatwijk