JvanKatwijk/dab-cmdline #78: Random "Sorry cannot handle service"

vale-max, 2021-05-12:
I compiled and run the examples 2 and 3 on a Raspberry Pi4 with a RTL SDR USB dongle.
Randomly they fail to start with "cannot handle service XY", sometimes at the immediately following attempt it works as expected, sometimes it fails various times in a row. It comes from the is_audioService check and when the error happens the service I'm trying to play is listed as part of the ensemble as normally.
Also sometimes the audio interrupts for few seconds.

On the very same hardware qt-dab works flawlessly, never experienced audio interruptions or failures to start a channel. The signal appears to be good, the dongle is attached to the TV VHF antenna and qt-dab details always show full green bars.

Any idea?

JvanKatwijk, 2021-05-12:
First of all:
the implementation of the fundamental algorithms is at least similar in
qt-dab and the dab library.
The difference is in the control.

Since the examples implement command line driven programs, there are two
time settings, one to basically check
that there is a form of a DAB signal, the second one, once it is concluded
that a DAB signal is most likely available,
as a timeout for the detection of an ensemble name.

These times can be set from the command line.

In my experience - with an SDRplay device - is that  the gain setting is
important.
When there are interruptions , it is usually due to weak signal

JvanKatwijk, 2021-05-12:
The quality of the signal decoding can be printed once every few seconds.
As the picture shows, three times 100 indicates good decoding,
if the first one is below 100, you definitely have a weak signal

vale-max, 2021-05-12:
I had already tried with longer tineouts and with autogain too, but it does not seem to change anything.

time ./dab-rtlsdr-2 -Q -D 20 -d 20 -C 12C -P "M DUE O"
dab_cmdline example II,
                        Copyright 2017 J van Katwijk, Lazy Chair Computing
options are T:D:d:M:B:P:O:A:C:G:p:Q
going for rtlsdr 227360000 50
OK, functions seem to be loaded
Found Rafael Micro R820T/2 tuner
samplerate set to 2048000
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
effective gain: gain 25.4
ALSA lib pcm_dsnoop.c:638:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5222:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5222:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib pcm_dsnoop.c:638:(snd_pcm_dsnoop_open) unable to open slave
connect(2) call to /dev/shm/jack-0/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
Hostapis: 2
Api 0 is ALSA
Api 1 is OSS
selecting device default
device bcm2835 HDMI 1: - (hw:0,0) seems available as 0
device bcm2835 Headphones: - (hw:1,0) seems available as 1
device SMSL USB AUDIO: Audio (hw:2,0) seems available as 2
device sysdefault seems available as 3
device front seems available as 4
device surround40 seems available as 5
device iec958 seems available as 6
device spdif seems available as 7
device default seems available as 8
device dmix seems available as 9
Suggested size for outputbuffer = 1280
Allocating 15 zero-copy buffers
ensemble * DAB Italia *   is (5009) recognized
program * RDS * Relax    is part of the ensemble
program 105 DAB          is part of the ensemble
program Deejay 30 Songs  is part of the ensemble
program Radio 24         is part of the ensemble
program * RDS *          is part of the ensemble
15:23
there might be a DAB signal here

we try to start program M DUE O
Because of is_audioservice: sorry  we cannot handle service M DUE O
program Radio Maria      is part of the ensemble
program RR News          is part of the ensemble
program Deejay           is part of the ensemble
program Kc1Test          is part of the ensemble
program Radio Radicale   is part of the ensemble
program Kc2Test          is part of the ensemble
program Kc3Test          is part of the ensemble
program RMALB.IA         is part of the ensemble
program Capital Funky    is part of the ensemble
program EPG              (data) is part of the ensemble
program R101             is part of the ensemble
program Capital          is part of the ensemble
program M DUE O DANCE    is part of the ensemble
program Radio 24 +1      is part of the ensemble
program M DUE O          is part of the ensemble
dab processor will stop

real    0m10.308s
user    0m2.000s
sys     0m0.214s

As you can see it didn't have to wait the timeout to sync and find the ensemble, nevertheless it reported the service as not possible to handle.

Just a few seconds later I restarted it with exactly the same command line and it worked, the signal quality is always very good, the lowest number I've seen is 93

we try to start program M DUE O
protection handler is eep_protection
we have now DAB+
        startaddress    = 720
        length          = 24
        subChId         = 18
        protection      = 2
        bitrate         = 32
msc quality = 98 100 100
msc quality = 100 100 100
msc quality = 100 100 100
msc quality = 100 100 100
msc quality = 100 100 100
msc quality = 100 100 100

vale-max, 2021-05-12:
I added the "because of is_audioservice" text as I had found the same "sorry" message in two different calls in the code

andimik, 2021-05-13:
@vale-max

Have you tried to use another gain setting instead of `-Q`?

Do you have RAI on 12B or 12D which is stronger than DAB Italia on 12C?

andimik, 2021-05-13:
This is strange

`-D 20 -d 20`

But 

```
real 0m10.308s
user 0m2.000s
```

So, you fear the options `-d` and `-D` are ignored?

JvanKatwijk, 2021-05-13:
It is always good to look at what the output is saying. It said that an
ensemble name was foiund within the time limites, but when the service
should start not all data
of the service is in.
That can of course be addressed by adding a sleep (whatever) in the code
after the code that verifies that the ensemble is recognized before
starting the service

While the 2016 DAB standard provides an option to check that descriptions
for all services of an ensemble are "in", I did not see so far any
transmission
providing the data for that option. So the current guess that we can
continue is to verify that the ensemble name is received

vale-max, 2021-05-13:
Yes, I've seen the "sorry cannot handle service" error on several programs on any channel which is tunable here and trying gain values from 10 to 90.
It is important to note that the audio is almost always ok, the interruptions happen very rarely, yesterday morning for example I listened perfectly to Radio2 on 12B and Virgin Radio on 12A for hours without issues, however starting the service sometimes requires two or three attempts.

Several times when the service fails to start the error message is displayed when the program list is still being completed and the name of the program I wanted is listed after the "we try to start program" message is logged.

Is it possible that the service start is attempted when the program list completion is still in progress and the program does not wait enough, despite the timeout settings?

JvanKatwijk, 2021-05-13:
That is what I was saying, the software triggers on "ensemble name found",
but that is as such no garantee that the descriptive data of all services
is read in. Since - in practice - it is not known when all descriptions of
all services are "in", I took the ensemblename as criterium.

You could add a "sleep (few seconds)" statement in after the detection of
the availability of an ensemblename to give the system - that is running in
the background - more time to ensure that the descriptions for the services
are delivered

gvanem, 2021-05-15:
Maybe it's related to this bug I found in `devices/rtlsdr-handler/rtlsdr-handler.cpp` while
playing with the Python module. 

A `python cmdline.py -g 100` reports:
```
Supported gain values (16): 0.0 3.4 6.8 10.2 13.7 17.1 20.7 24.0 27.8 31.2 34.6 38.2 41.6 45.3 48.8 52.7
effective gain: gain -188739530.-7
```

Seems to be an off-by-1 index-error in `rtlsdrHandler::rtlsdrHandler()` which I fixed by:
```diff
--- a/devices/rtlsdr-handler/rtlsdr-handler.cpp 2021-04-25 12:42:53
+++ b/devices/rtlsdr-handler/rtlsdr-handler.cpp 2021-05-15 13:14:55
@@ -178,10 +178,12 @@
        if (autogain)
           rtlsdr_set_agc_mode (device, 1);
        (void)(this -> rtlsdr_set_center_freq (device, frequency));
+
+       int16_t _gainsCount = gainsCount - 1;
        fprintf (stderr, "effective gain: gain %d.%d\n",
-                                     gains [theGain * gainsCount / 100] / 10,
-                                     gains [theGain * gainsCount / 100] % 10);
-       rtlsdr_set_tuner_gain (device, gains [theGain * gainsCount / 100]);
+                                     gains [theGain * _gainsCount / 100] / 10,
+                                     gains [theGain * _gainsCount / 100] % 10);
+       rtlsdr_set_tuner_gain (device, gains [theGain * _gainsCount / 100]);

        if ( this       -> deviceOptions && rtlsdr_set_opt_string )
                rtlsdr_set_opt_string(device, deviceOptions, 1);
```

vale-max, 2021-05-17:
Hi @JvanKatwijk 
with the sleep added it works perfectly, 100% working at first attempt.

I increased the sleep to 4 at line 471 of dab-scanner/main.cpp too as it was often missing services.

Still in dab-scanner, I also modified the layout of the service lines to include the channel and a leading field that indicates whether it is an audio or data service, just to make life easier when parsing the channel list.

Now a service line looks like:
`audioservice;BBC WorldService;12A;C238;7;528;42;56;DAB+;1;EEP 3-A;1/2;English;
`

I pushed my changes to my fork of your repository, to eventually merge them in case you consider them useful.

Thanks a lot for this wonderful piece of software.

closed 2021-05-17 by vale-max

JvanKatwijk, 2021-05-18:
Thanks

I see the point. Well, the whole idea of the timeSyncSet variable is to
speed up if no signal can be detected.
removing that test in the loop is probably best. The duration of that loop
is set by timesynctime, ususally about 5 seconds should be sufficient

vale-max, 2021-05-18:
In the optic of speeding up the scan, it maybe would be useful to have the possibility to provide a list of the channels to be scanned rather than just the starting channel.
In my area only 5 channels have a signal, nevertheless the scanner goes through the whole spectrum even if I start from the first available (12A, and last channel with a signal is 13F)

JvanKatwijk, 2021-05-18:
yes,
last september there was a question about specifying a last of channels. I
made a separate version - obviously based on the dab-library - with which
you specify the channels to visit. Look at
https://github.com/JvanKatwijk/channel-scanner