JvanKatwijk/dab-cmdline #79: no audio out with Raspberry Pi OS Lite

sulmfish, 2021-05-19:
I installed dab-cmdline example 2 on a Raspberry Pi 3B with Raspberry Pi OS Lite (headless) as described in the related documentation. I got it working but there is no audio out. I can generate an audio file with option -O, but it sounds horrible (at least with omxplayer). But no audio out with OS Lite and from remote terminal.
If I install it based on the Raspberry Pi Desktop OS and start example 2 from Desktop, everything is fine. Installation via remote terminal is exactly the same.
Do I have to install any additional libraries? Any idea?

JvanKatwijk, 2021-05-19:
Do I understand it correctly that in case you install the software on a
headless installed RPI there is no sound, while when you
install the software directly (i.e. from a connected terminal) everything
works fine?

If you install the software remotely, and you start the compiled
executable, you do get response from the program stating that there are
services etc,
but whatever soundcard you select on the RPI there is no sound on the
audioplug?

sulmfish, 2021-05-19:
I found the solution: In the standard Raspberry Pi OS (with desktop) pulseaudio is installed and is started automatically after boot. In the Rapsberry Pi OS Lite (which is without desktop) pulseaudio is not installed. So "sudo apt-get install pulseaudio"  is the solution and pulseaudio must be started before dab-cmdline by "pulseaudio --daemonize" for example.

closed 2021-05-19 by sulmfish

andimik, 2021-05-20:
@sulmfish 

Good hint. Maybe it's worth to mention it in the Readme.

sulmfish, 2021-05-20:
By the way: I didn't need "sudo apt-get install mesa-common-dev" and "sudo apt-get install libgl1-mesa-dev". I think these are installed with pulseaudio.

sulmfish, 2021-05-20:
One thing to mention belonging to Raspberry Pi OS Lite: I had to "sudo dpkg-reconfigure locales". I always selected en_GB.UTF8,  en_US.UTF8 and de_DE.UTF8. Probably too much, but it worked.
This can be done also via raspi-config.