JvanKatwijk/dab-cmdline #81: Example 1 - dataOutHandler question

Morrisman68, 2021-12-28:
I am trying to work from a known basis to extract 'Now Playing information from BBC Radio 3 DAB and having problems. (I am working from an example as my coding skills are not great, nor my understanding of DAB. I am very grateful for the code and the hard work behind it - thank you!)

My modification looks like this:

//	The function is called from within the library with
//	a string, the so-called dynamic label
static
void	dataOut_Handler (std::string dynamicLabel, void *ctx) {
	(void)ctx;
	fprintf (stderr, "data out called\n");
	fprintf (stderr, "%s\n", dynamicLabel. c_str ());
}
//
The only output produced is typically of this form:

data out called
NlaLuur)

Observation of a DAB tuner shows that the program data being streamed bears no resemblance whatsoever to this, and I am at a loss to understand what is happening, unless I am perhaps mis-handling a pointer?

All suggestions would be welcome!

Many thanks,

Ben

JvanKatwijk, 2021-12-28:
Dear Ben

Assuming you are running example 2. Some time ago the interface was changed
to provide a more C-like (and less C++ like)
interface, so dataOut is defined as given below,
//      dynamic label data, embedded in the audio stream, is sent as string
        typedef void (*dataOut_t)(const char *, void *);

it asks for a char * as parameter rather than a std::string.

/ /     The function is called from within the library with
//      a string, the so-called dynamic label
static
void    dataOut_Handler (const char * dynamicLabel, void *ctx) {
        (void)ctx;
        std::cerr << std::string (dynamicLabel) << "\r";
}

Morrisman68, 2021-12-28:
Many thanks, Jan.

I am slightly puzzled by the fact that I pulled -0.8 as the latest version - am I missing something? I am a stranger to github, too!

Ben

JvanKatwijk, 2021-12-28:
which of the examples you are using, just example 2?

JvanKatwijk, 2021-12-28:
Now I see the title, it is much easier to use example 2, since then the
library functions are "comoiled in",
Last summer there were - based on some requests quite some changes, so my
advice is to delete the while
cmdline directory, and load a fresh one

git clone https://github.com.JvanKatwijk/dab-cmdline

then
cd dab-cmdline/example-2

then mkdir build
cd build
cmake .. -DRTLSDR

Morrisman68, 2022-01-03:
Jan, 

all is now hummin' and tickin' just as I have been hoping - many thanks for your work, sharing and kind help.

Heel hartelijk bedankt!

Ben

closed 2022-01-03 by Morrisman68

JvanKatwijk, 2022-01-03:
do   not hesitate to contact me if  there are questions

all the best
jan