JvanKatwijk/dab-cmdline #83: Example 3 crashes

vale-max, 2022-06-10:
Hello Jvan,
dab-example-3, compiled from my fork of your repo, is randomly crashing at startup:
Command used is:
/dab-rtlsdr-3 -S 5241 -C 12A >/dev/null

```
we try to start program Virgin Radio
protection handler is eep_protection
we have now DAB+
        startaddress    = 690
        length          = 36
        subChId         = 19
        protection      = 2
        bitrate         = 48
terminate called without an active exception
Aborted (core dumped)

```
most of the times it works, sometimes crashes.
The stack backtrace is:
```
#0  __pthread_kill_implementation (threadid=547914510496, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x0000007f9e2327e4 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x0000007f9e1eb6fc in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x0000007f9e1d78b0 in __GI_abort () at abort.c:79
#4  0x0000007f9e4f32d4 in __gnu_cxx::__verbose_terminate_handler () at /usr/src/debug/gcc/libstdc++-v3/libsupc++/vterminate.cc:95
#5  0x0000007f9e4f065c in __cxxabiv1::__terminate (handler=<optimized out>) at /usr/src/debug/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48
#6  0x0000007f9e4f06e0 in std::terminate () at /usr/src/debug/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:58
#7  0x0000005577c551d8 in std::__terminate () at /usr/include/c++/12.1.0/aarch64-unknown-linux-gnu/bits/c++config.h:312
#8  std::thread::operator= (__t=..., this=<optimized out>) at /usr/include/c++/12.1.0/bits/std_thread.h:164
#9  streamer::restart (this=0x7f9c20c010) at /root/dab-cmdline/example-3/streamer.cpp:58
#10 0x0000005577c54080 in pcmHandler (buffer=0x7f9be54100, size=3840, rate=<optimized out>, isStereo=<optimized out>, ctx=<optimized out>)
    at /root/dab-cmdline/example-3/main.cpp:200
#11 0x0000005577c7ca6c in faadDecoder::output (rate=48000, isStereo=<optimized out>, size=3840, buffer=<optimized out>, this=0x7f988027a0)
    at /root/dab-cmdline/example-3/../library/includes/backend/audio/faad-decoder.h:56
#12 faadDecoder::MP42PCM (this=this@entry=0x7f988027a0, sp=sp@entry=0x7f923fdc60,
    buffer=buffer@entry=0x7f923fdde0 "\200)\030I\233\363\034\236\254i\237\321\262\365,Y!\376nD\221O8\321\035\333v<:", bufferLength=bufferLength@entry=203)
    at /root/dab-cmdline/example-3/../library/includes/backend/audio/faad-decoder.h:169
#13 0x0000005577c7a0ec in mp4Processor::processSuperframe (this=this@entry=0x7f98802480, frameBytes=<optimized out>, base=<optimized out>)
    at /root/dab-cmdline/library/src/backend/audio/mp4processor.cpp:333
#14 0x0000005577c7af10 in mp4Processor::addtoFrame (this=0x7f98802480, V=<optimized out>) at /root/dab-cmdline/library/src/backend/audio/mp4processor.cpp:177
#15 0x0000005577c75f8c in audioBackend::processSegment (this=this@entry=0x7f9a802a80, Data=<optimized out>) at /root/dab-cmdline/library/src/backend/audio-backend.cpp:162
#16 0x0000005577c761f8 in audioBackend::run (this=0x7f9a802a80) at /root/dab-cmdline/library/src/backend/audio-backend.cpp:171
#17 0x0000007f9e5278ac in std::execute_native_thread_routine (__p=0x7f9c20ea70) at /usr/src/debug/gcc/libstdc++-v3/src/c++11/thread.cc:82
#18 0x0000007f9e230aec in start_thread (arg=0x7fd0e21bdf) at pthread_create.c:442
#19 0x0000007f9e29a5dc in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:79

```

JvanKatwijk, 2022-06-10:
> #0  __pthread_kill_implementation (threadid=547914510496, ***@***.***=6, ***@***.***=0) at pthread_kill.c:44
> #2  0x0000007f9e1eb6fc in __GI_raise ***@***.***=6) at ../sysdeps/posix/raise.c:26
> #12 faadDecoder::MP42PCM ***@***.***=0x7f988027a0, ***@***.***=0x7f923fdc60,
> #13 0x0000005577c7a0ec in mp4Processor::processSuperframe ***@***.***=0x7f98802480, frameBytes=<optimized out>, base=<optimized out>)
> #15 0x0000005577c75f8c in audioBackend::processSegment ***@***.***=0x7f9a802a80, Data=<optimized out>) at /root/dab-cmdline/library/src/backend/audio-backend.cpp:162

Looking at the trace the error occurs in the "streamer". Now looking at the
streamer I see that the segment size is 2 * 4800 int's,
but the delay in the loop is 10 microseconds, which seems to me a little
short since the samplerate is 48000
You need the streamer?

vale-max, 2022-06-11:
I'm not sure  what the streamer does, I use the code to output to a pipe that at the other end has ffmpeg processing it and spitting it out in aac and rtsp format to an rtsp server

```
dab-rtlsdr-3 -S $1 -C $2 > $MYPIPE &
DABPID=$!
ffmpeg -re -stream_loop -1 -ar 48000 -ac 2 -f s16le  -i $MYPIPE -vn -c:a aac -b:a 160k -metadata title="Dab radio" -f rtsp rtsp://localhost:$3/$4 >/dev/null 2>&1 &
```

JvanKatwijk, 2022-06-11:
As far as I recall, the streamer was added to fill up gaps in the output
stream, if the incoming data stops, the output
continues with 0 samples.
I did a small test this morning with the period in the streamer task set to
0.1 second rather than 0.01 second
and so far there were no problems

But then, without the change I did not have a crash as well (but only a few
short tests)

vale-max, 2022-06-11:
I tried to compile example-3 commenting out "add_definitions (-DSTREAMER_OUTPUT)" in CMakeLists.txt, I hope I understood correctly how to disable the streamer.
It didn't crash so far and the audio seems to work. I guess it would notice some difference in case of weak signals, right?

JvanKatwijk, 2022-06-13:
Sorry for the delay in answering, been a little budy

The streamer was - as said - basically a buffer handling gaps in the audio
output. The idea is to have a continuous stream of data
regardless of whether or not the DAB software has output
If the program you are feeding the output in can handle gaps, then the
streamer has no use.

Saturday I ran some examples with the period in the streamer set to 100
msec (4800 samples), and
for about 5 minutes it ran without problems

vale-max, 2022-06-14:
For me it never crashed once it started playing sound, just at the very
beginning before any sound is heard.
I'm using the supposedly-not-having-the-streamer version and it did not
crash until now

vale-max, 2022-06-17:
Another thing I noticed is that in the version compiled with the streamer the audio starts, then there's a moment of silence an then it starts playing uninterrupted (and keeps playing for hours always, no crashes once it starts)

The version compiled without the streamer starts reproducing audio without the short interruption at the beginning

When the version with streamer crashes it starts before reproducing anything.

I did the tests piping the output to aplay:
dab-rtlsdr-3 -S 5241 -C 12A|aplay -D default:CARD=AUDIO -c 2 -f S16_LE -r 48000 -

JvanKatwijk, 2022-06-17:
Based om what you are saying I would expect that the streamer is the
culprit.
If it crashes before any output, it looks like an uninitlaized variable. In
my system all locals are automatically
set to 0, but the weekend I'll have a good look at the code

JvanKatwijk, 2022-06-17:
Just to be certain:
The crash happens sometimes, i.e. not always, but if it happens it is just
after starting up, including the streamer.
If no streamer is in the configuration, there is a few seconds no siund,
but no crash

vale-max, 2022-06-17:
The crash happens only when the streamer is enabled in the configuration. Happens sometimes, i.e. not always, but if it happens it is just after starting up, before ANY sound is heard.
When the streamer is in the configuration and it does not crash, when it starts it outputs a moment of audio, then a moment of silence then the audio starts and goes on playing without issues.

When the streamer is not configured at compile time the example never crashes and there is no moment of silence after sound starts.

I tried to produce an audio sample writing to file, but the sample produced by the version with the streamer is sound-silence-sound while when the output is piped to aplay or to the rtsp-simple-server the audio is ok

[soundsamples.zip](https://github.com/JvanKatwijk/dab-cmdline/files/8929556/soundsamples.zip)

JvanKatwijk, 2022-06-18:
Interesting issue

I ran example 3 dozens times, sometimes for a few minutes, sometimes for
app 10 minutes without any issue
Looking at the symptons you describe, it is most likely some variable that
was uninitialized or a time issue
wrt variables: I had a good look and could not detect uninitialized
variables,
wrt timing: I changed the streamer in that it now starts on creation of the
instance

Be aware that the audio samples are not "wav" samples

If you run with the address sanitizer linked in and get a crash I am very
interested in the output

vale-max, 2022-06-20:
Hello Jvan,
I tried the new code with streamer enabled, started it several times and it never crashed, also if I redirect to file the audio has about 6 seconds of silence at the beginning (which I guess is the time it takes to initialize all the SDR stuff) but then it is ok, not fragmented as the previous version

JvanKatwijk, 2022-06-20:
OK
Thanks for the report. What I changed was starting the thread implementing
the buffering earlier, so it must have
been an issue with timing, i.e. calling on a function before the task
actually ran

best
jan