JvanKatwijk/dab-cmdline #86: dab-library within MPD plugin

macmpi, 2022-09-06:
Hi,
As [MPD](https://www.musicpd.org/) is one of the most versatile audio player solution, I was wondering if there is any plan to build some sort of [MPD source plugin](https://mpd.readthedocs.io/en/latest/plugins.html#input-plugins) so that DAB/DAB+ can be smoothly integrated in many MPD-based solutions.
Thoughts?

JvanKatwijk, 2022-09-06:
Well, nothing is impossible, but I am not familiar with mdp. So, does it
take aac frames or audio samples as input, if so, it is quite
straightforward since variants that generate that type of output exist

macmpi, 2022-09-06:
Yes MPD deals with most formats, including plain samples.
It does manages playlists also and all sort of tags/metadata, so scanned DAB radio channels could easily be filled in such playlist to that any MPD client (GUI / cmdline) would seamlessly handle channel selection.
There is a huge range of [GUI clients](https://www.musicpd.org/clients/) available on any platforms.

vale-max, 2023-02-16:
@Lamarqe  Are you sure the audio is decoded correctly, e.g. the pitch is correct, or the tempo?
As far as I remember MPD uses ffmpeg to decode the audio, and ffmpeg does not support correctly the DAB+ AAC frames that have 960 rather than 1024 samples.
I tried repeatedly to have this issue fixed in ffmpeg but for some reason this bug keeps being ignored:
https://trac.ffmpeg.org/ticket/1407

Lamarqe, 2023-02-17:
Yes, I am. I use this setup every day. 😄 

I had initial problems with the tempo when I used example-3. I tried to work around with some ffmpeg parameter to manually correct it, but the results were unacceptable.

As soon as I switchted to example 4 (_"No sound decoding takes place. The MP2 frames (in case of DAB) and the AAC frames (in case of DAB+) are just emitted into a file or to stdout."_), the problems disappeared.

Lamarqe, 2023-02-17:
One more comment: 
Example 4 avoids and re-encoding withing dab-cmdline
The ffmpeg command line in my tvheadend helper script also does not do any re-encoding. The only thing it does, is to wrap the dab-cmdline stream into an mpegts envelope. This is the prerequisite for tvheadend which does the http serving (which I then again use in mpd as a stream URL). Bottom line: No re-encoding at all. At the end, mpd streams the original DAB+ stream (in my case via pulseaudio to chromecast and chromecast does the decoding).

Lamarqe, 2023-11-25:
@vale-max I recently changed my setup and then also faced issues with streams that could not be processed due to the mentioned ffmeg incompatibility.

This ended up in replacing the ffmpeg command line in my helper script with a VLC one:
```
#!/bin/bash
/usr/bin/dab-rtlsdr-4 -M 1 -B "BAND III" -C $1 -P "$2" -G 80 2>/dev/null | cvlc - --sout "#transcode{acodec=mpga,channels=2,rate=44100,ab=320}:std{access=file,mux=ts,tsid=1,dst=-}" -q --intf dummy vlc:quit 
```
So instead of handing over the stream as-is to tvheadend, it is getting processed and converted before by VLC. Apparently, libVLC does not have this problem.

Lamarqe, 2024-08-05:
I implemented an application which can be integrated with MPD: https://github.com/Lamarqe/mpdcast-dab