JvanKatwijk/dab-cmdline #89: tii_data_t doesn't include userData parameter

srcejon, 2023-02-07:
Most of the callbacks in API_struct include the userData parameter. However, tii_data_t doesn't, which is problematic if trying to receive from multiple transmitters.

srcejon, 2023-02-07:
The additional parameter would be userData that is passed to dabInit, the same that is passed to all of the other callbacks. 

So the call would be changed to:

    show_tii (res, userData);

JvanKatwijk, 2023-02-08:
done, I did not look (yet?) to the python example though

srcejon, 2023-02-10:
Thanks.

Just to clarify it's usage, we are plotting the location of the transmitter on a map. So when decoding multiple DAB streams, the user data is used to indicate which stream a TII is for.

closed 2023-02-10 by srcejon