JvanKatwijk/dab-cmdline #92: Problem in example 1 and 2

1337Misom, 2023-05-14:
Hi,
I just wanted to mention you forgot to call `set_audioChannel(theRadio,&ad);` after `dataforAudioService()` in examples 1 and 2. Without it doesn't start the decoding process (at least with an rtlsdr).

JvanKatwijk, 2023-05-15:
As far as I  can see, the code reads
 if (run. load ()) {
           dataforAudioService (theRadio,
                             programName. c_str (), &ad, 0);
           if (!ad. defined) {
              std::cerr << "sorry  we cannot handle service " <<
                                                 programName << "\n";
              run. store (false);
           }
        }

        if (run. load ()) {
           dabReset_msc (theRadio);
           set_audioChannel (theRadio, &ad);
        }

While the code guarded by the second condition could have been merged with
the code guarded byh the first
condition, it is clear that if (a) "run" is set to true, and the service is
defined, the function "set_audioChannel"
is called.
So I am a little cofused by your statment, I just ran a test and it runs
fine, with sound

JvanKatwijk, 2023-05-15:
> The code you send would work ,but the one currently in the git doesn't
> include the if statement or any other form of set_audioChannel

Then ther eis some miscommunication, this is the code - just extracted -
from example 2 ikn the repository
What code do you have, and whgere/how did you get/extract it

if (run. load ()) {
dataforAudioService (theRadio,
programName. c_str (), &ad, 0);
if (!ad. defined) {
std::cerr << "sorry we cannot handle service " <<
programName << "\n";
run. store (false);
}
}

if (run. load ()) {
dabReset_msc (theRadio);
set_audioChannel (theRadio, &ad);
}

while (run. load () && (theDuration != 0)) {
if (theDuration > 0)
theDuration --;
sleep (1);
}

JvanKatwijk, 2023-05-15:
> Okay then it only seems to be a problem in example 2

You were right aaout example-1, it is adapted
In my defence I can say that the examples are - as mentioned in different
places -
*      E X A M P L E  P R O G R A M
 *      This program might (or might not) be used to mould the interface to
 *      your wishes. Do not take it as a definitive and "ready" program
 *      for the DAB-library

anyway, thanlks for the tip and good luck