JvanKatwijk/dab-cmdline #93: Formal things

1337Misom, 2023-05-15:
Hi,
After looking around in your code i noticed you used throw and catch with different integers and printing the errors directly into stderr. Wouldn't it be more practical to actually define exceptions which will then be thrown? Because if you were just too lazy i would maybe start creating exceptions. Another question also is do you use any code formatter for the code or is it just your code style?

JvanKatwijk, 2023-05-16:
well,
the "library approach" has a history. About 6 (maybe 7) years ago, there
were some people asking for a command line rather than a graphical
interface. So, about 6 years ago, I took the - then existing - sources of
qt-dab and removed the dependencies on Qt.
In 2 or 3 iterations there was the library. Then people started asking on
how to use it, so I made some example programs,
in the end even with a python interface. I made essentially only one larger
change to the library, since then, rather than having to submit a number
of callback  functions as separate parameters, I created a single structure
to be passed as parameter.
(I added some support for e.g. Pluto and line devices, but the device
support is almost the same as for
e.g. the eti  handler, and for some other local programs,

So, the code seems to work, but  can be considered "old-fashioned".
One of the more serious things that is missing is the handling of changes
in the configuration, the BBC seems to apply quite a number
of changes in the configuration dynamically. (the same applied to the "eti"
handler, that software is from about the same period).

Wrt to the code srtyle
I am not completely consistent with the style (especially naming of
entities is not consistent over time.
The lay out is however, especially spacing is impotant for readability

Wrt exceptions
One of the few places (I think) where I use exceptions is with the
instantiations of a device handler.
As far as I can see, the most important message from this kind of
exceptions is: be careful, the device interface does not work
In some other software I map (for some devices) the number returned to a
real error message describing the error.

best
jan

1337Misom, 2023-05-16:
Ok so i wil now add specific exceptions into my fork and if you want i can then create a pr with the changes.

1337Misom, 2023-05-17:
I now have added simple exceptions and fixed a typo in my fork. If you are ok with the changes tell me so that i can create a pr.

JvanKatwijk, 2023-05-18:
apology for the delay. Youn are always welcome to do a PR
best
jan

closed 2023-05-18 by 1337Misom