JvanKatwijk/dab-cmdline #100: ppa for dab-cmdline

Lamarqe, 2023-11-15:
I currently prepare a ppa for dab-cmdline: https://launchpad.net/~lamarqe/+archive/ubuntu/ppa

@JvanKatwijk : Would be willing to take over the respective debian configuration files when I submit them via a MR?
This would allow me to create the builds directly from the main repo instead of my personal fork.

Currently I have all RTLSDR builds enabled and running. This is easy as rtlsdr has its own apt packages by now. 
Preferably, I would hand over this status with a MR.

Afterwards, I would need to check if and how also the other build types could be enabled. 
Not sure if I can download source code from the internet in the launchpad containers.

JvanKatwijk, 2023-11-25:
Maybe my ignocance but I do not understand what you are saying, for me a
ppa is a pulsed power amplifier, and an MR is a dutch
abbreviation for "MedezeggenschapsRaad", and I am quite certain that that
is not what you try to say

jan

Lamarqe, 2023-11-25:
I noticed that I might not have been specific enough and already had adjusted the subject to make it more clear.

It is about packaging dab-cmdline into a debian package.
I would like to merge the respective build instruction to dab-cmdline.

Having that, it is possible to configure launchpad.net to automatically created an up-to-date apt package whenever there are new commits to the repo here at github. 

So everybody can simply do "apt install dab-cmdline" instead of having to compile the package themselves.
I have already configured a respective apt repo over there. This is already publicly available for everybody since recently. 
But its based on my github for (which contains the build instruction)

Having the build instruction merged to your repo would allow me to point the my launchpad config directly your main repo.

Lamarqe, 2024-01-01:
Closed as the maintainer is not interested in it. I will keep it in my personal repo.

closed 2024-01-01 by Lamarqe