JvanKatwijk/dab-cmdline #104: New version can't decode what an older version is able to

montgomeryb, 2024-11-12:
I have been using dab-cmdline on my raspberry 3B for quite a while on 32 bit debian buster with a sdr usb stick. I upgraded to 64 bit bookworm. I tried updating the libraries but with the latest version I cannot pickup any of the stations. If I use the version I had from several years ago, it works (most of the time).

I'm not sure if there's anything I can do to try and track down the issue?

```
root@water:/home/bryan/src2/rtl-sdr/build# /home/bryan/src2/dab-cmdline/example-3/build/dab
-rtlsdr-3 -Q on -C 11C -P "BBC Radio Solent" >/dev/null
dab_cmdline example III,
                          Copyright 2017 J van Katwijk, Lazy Chair Computing
options are i:T:D:d:M:B:P:O:A:C:G:p:Q:S:
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
there might be a DAB signal here
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping 1

no ensemble data found, fatal
dab processor will stop
root@water:/home/bryan/src2/rtl-sdr/build# dab-rtlsdr-3.old -Q on -C 11C -P "BBC Radio Sole
nt" >/dev/null
dab_cmdline example III,
                          Copyright 2017 J van Katwijk, Lazy Chair Computing
options are q:T:D:d:M:B:P:O:A:C:G:p:Q
going for rtlsdr 220352000 50
OK, functions seem to be loaded
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
samplerate set to 2048000
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9
25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
effective gain: gain 25.4
there might be a DAB signal here
Capital S Coast (C870) is part of the ensemble
HITS RADIO S CST (C1E2) is part of the ensemble
ensemble South Hampshire is (C1A3) recognized
Heart Hampshire (C365) is part of the ensemble
Angel Vintage (CDC2) is part of the ensemble
V2 Radio (CDEE) is part of the ensemble
Grt Hits S Coast (C3AD) is part of the ensemble
Station Logos (data) (E1C000CA) is part of the ensemble
Smooth Hampshire (CFD2) is part of the ensemble
Nation Radio (C2C4) is part of the ensemble
Easy Radio (CEF0) is part of the ensemble
HOT GOLD (CAED) is part of the ensemble
Station Logos (data) (E1C000CA) is part of the ensemble
BBC Radio Solent (C932) is part of the ensemble

we try to start program BBC Radio Solent
protection handler is uep_protection
we have now DAB
        startaddress    = 288
        length          = 96
        subChId         = 4
        protection      = 3
        bitrate         = 128
^Cu're listening to BBC Radio Solentic you love
```

JvanKatwijk, 2024-11-13:
When I run with a similar configuration on example 3, i.e. an RTLSDR
device, I get a list of gain values,
Found Fitipower FC0013 tuner
samplerate set to 2048000
Supported gain values (23): -9.-9 -7.-3 -6.-5 -6.-3 -6.0 -5.-8 -5.-4 5.8
6.1 6.3 6.5 6.7 6.8 7.0 7.1 17.9 18.1 18.2 18.4 18.6 18.8 19.1 19.7
effective gain: gain 18.8
there might be a DAB signal here

Since I do not have a decent antenna that fits on the connector I do not
get an ensemble, but reading the code, tou should get some output there
then at least you can see the actual gain
best
jan

montgomeryb, 2024-12-10:
Hello, I did browse the code but nothing jumped out at me.
I did notice though on the older version it looks like the gain was printed to stderr , but on the newer version it seems to be printed on stdout.

The best I could get with the latest code is:

```
./dab-rtlsdr-3  -G 75 -C 11C -P "BBC Radio Solent"
dab_cmdline example III,
                          Copyright 2017 J van Katwijk, Lazy Chair Computing
options are i:T:D:d:M:B:P:O:A:C:G:p:Q:S:
going for rtlsdr 220352000 75
OK, functions seem to be loaded
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
samplerate set to 2048000
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
effective gain: gain 38.6
there might be a DAB signal here
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping 1

no ensemble data found, fatal
dab processor will stop

```

Additionally, on occasion I would get:
```=================================================================
==9664==ERROR: AddressSanitizer: alloc-dealloc-mismatch (operator new [] vs operator delete) on 0x007fa4e94d00
    #0 0x7faadbcab8 in operator delete(void*) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:152
    #1 0x5581010fdc in interLeaver::~interLeaver() (/usr/local/bin/dab-rtlsdr-3+0x30fdc)
    #2 0x558100df64 in ofdmDecoder::~ofdmDecoder() (/usr/local/bin/dab-rtlsdr-3+0x2df64)
    #3 0x558100bc2c in dabProcessor::~dabProcessor() (/usr/local/bin/dab-rtlsdr-3+0x2bc2c)
    #4 0x5580ffeb2c in main (/usr/local/bin/dab-rtlsdr-3+0x1eb2c)
    #5 0x7faa6a773c in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #6 0x7faa6a7814 in __libc_start_main_impl ../csu/libc-start.c:360
    #7 0x55810005ac in _start (/usr/local/bin/dab-rtlsdr-3+0x205ac)

0x007fa4e94d00 is located 0 bytes inside of 4096-byte region [0x007fa4e94d00,0x007fa4e95d00)
allocated by thread T0 here:
    #0 0x7faadbc2b4 in operator new[](unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:98
    #1 0x5581010e0c in interLeaver::interLeaver(unsigned char) (/usr/local/bin/dab-rtlsdr-3+0x30e0c)
    #2 0x558100ea4c in ofdmDecoder::ofdmDecoder(unsigned char, RingBuffer<std::complex<float> >*) (/usr/local/bin/dab-rtlsdr-3+0x2ea4c)
    #3 0x558100be5c in dabProcessor::dabProcessor(deviceHandler*, API_struct*, RingBuffer<std::complex<float> >*, RingBuffer<std::complex<float> >*, void*) (/usr/local/bin/dab-rtlsdr-3+0x2be5c)
    #4 0x5581008a78 in dabInit (/usr/local/bin/dab-rtlsdr-3+0x28a78)
    #5 0x5580ffdfb8 in main (/usr/local/bin/dab-rtlsdr-3+0x1dfb8)
    #6 0x7faa6a773c in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #7 0x7faa6a7814 in __libc_start_main_impl ../csu/libc-start.c:360
    #8 0x55810005ac in _start (/usr/local/bin/dab-rtlsdr-3+0x205ac)

SUMMARY: AddressSanitizer: alloc-dealloc-mismatch ../../../../src/libsanitizer/asan/asan_new_delete.cpp:152 in operator delete(void*)
==9664==HINT: if you don't care about these errors you may set ASAN_OPTIONS=alloc_dealloc_mismatch=0
==9664==ABORTING
```

JvanKatwijk, 2024-12-12:
Right now I do not have a working RTL stick, they are mechanically a little
weak, and probabcly some connection internally broke

Witj the latest version - using a SDRplay device - all orks fine, at least
on my laptop.
I'll have to dust off my RPI 3B and install bookwork. Give me a few days
for the update and then I can report on the functioning with my RSP

JvanKatwijk, 2024-12-15:
Having installed bookworm, I got some problems with the installation of the
libraries required for the dab-cmdline examples,
while cmake states that pacjages are being installed, the libraries are
apparently not detected and no makefile is generated
I am not very strong with Cmake (better: fairly weal), so if you can tell
me how you installed the packages I can continue on the RPI3

best
jan

montgomeryb, 2024-12-17:
Hi Jan,
Thanks again for digging into this. With the 1GB Pi 3B, when I was running the 32 bit buster I had issues with CMake, I think in particular with fftw3. Now on the 64 bit readme, I can build following the debian instructions on the README page. 
However it seemed that memory use was carzy and some of the packages for the pi are so old that I actually built fftw and faad from source. Now when I run ` cmake .. -DRTLSDR=on`
it completes, but gives me a warning 
```
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1")
CMake Warning (dev) at /usr/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (FFTW3f).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.25/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  cmake/Modules/FindFFTW3f.cmake:6 (INCLUDE)
  CMakeLists.txt:141 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.10
CMake Warning (dev) at /usr/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (FFTW3F)
  does not match the name of the calling package (FFTW3f).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/Modules/FindFFTW3f.cmake:40 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:141 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.
```
I doubt I can offer much help, but feel free to ask.

JvanKatwijk, 2024-12-19:
apologies for the late reply. I am a little busy/
Next weekend I'll give it another try.

One thing though,, did you try the autoconf?
As said, I do not have a (reliable) stick right now, but with the sdrplay I
did not have problems, The phenomenon uou sketched looked like a gain
issue, since the
software does time synchronization (which is relatively easy), and fails
wiith detecting data.

As said, I'll give it another try this weekend

besy
jan

JvanKatwijk, 2024-12-21:
Well,
I did a test today
dirst I created a 32 bit bookworm and managed to get dab-cmdline example 2
running. Since I use an older usb stick as root device that
apaprently gave some hickups with readong and writing, it took a while, but
example-2 ran with an SDRplay RSP device
Then I installed the 64 bit version and behavious was exactly the same

So your problem is most likely a gain setting issue for the stick you are
usingm there are no observable differences in behavious between
the 32 and 64 bit version of bookworm (I did not look at the processorload)

best
jan

\

montgomeryb, 2025-03-16:
I recently upgraded the raspberry from model 3 to 4. Works fine now - go figure. Thanks again for looking at this.

closed 2025-03-16 by montgomeryb