JvanKatwijk/dab-cmdline #107: Dab-Scanner: -Dd and -d parameters are ignored

jeremyp3, 2025-06-25:
i had already noticed this some time ago but hadn't taken the time to make an issue, it's now done.

using dab-scanner, i notice that the -d and/or -D parameters are not taken into account. whatever value i set, whether for the detection of a dab signal (-d) or for the detection of the complete set (-D), the value is not taken into account ...

is it possible to correct the bug? :)

thanks !

jeremyp3, 2025-06-26:
then yes it builds, but it doesn't recognize the option :( i think you have to change it somewhere else than in line 267 of main.cpp :)

thanks for your work and sorry for the comment spam!

jeremyp3, 2025-06-26:
here, it doesn't seem to work. to give more info on my config:

* debian testing x86_64

* SDR key
cmake version 3.31.6

here's the command output, if it helps with debugging

:~ $ /tmp/dab-scanner-rtlsdr -C9A -I9
dab_scanner V 2.0alfa,
[...]
going for rtlsdr 202928000 50
OK, functions seem to be loaded
[...]

checking data in channel 9A
ensemble PAU 9A is (F05A) recognized

Ensemble PAU 9A; ensembleId F05A; channel 9A; 

Audio services
type;program name;channel;serviceId;subchannelId;start address;length (CU); bit rate;DAB/DAB+; genre; prot level; code rate; language

audioservice;100/100         ;9A;F692;0;0;66;88;DAB+;0;EEP 3-A;1/2;French;
audioservice;ATOMIC          ;9A;F6A8;1;66;66;88;DAB+;0;EEP 3-A;1/2;French;
audioservice;WIT FM dab+     ;9A;FD34;3;198;66;88;DAB+;0;EEP 3-A;1/2;French;
audioservice;CHANTE FRANCE   ;9A;FE43;4;264;66;88;DAB+;0;EEP 3-A;1/2;French;
audioservice;NRJ PYRENEES    ;9A;F220;6;396;66;88;DAB+;0;EEP 3-A;1/2;French;
audioservice;7RADIO  DAB+    ;9A;FDC1;7;462;66;88;DAB+;0;EEP 3-A;1/2;French;
audioservice;Pyrénées FM   ;9A;F64F;8;528;66;88;DAB+;0;EEP 3-A;1/2;French;
audioservice;R BONNE NOUVELLE;9A;FD41;9;594;66;88;DAB+;0;EEP 3-A;1/2;French;
audioservice;RADIO MARIAFR   ;9A;F9F4;10;660;66;88;DAB+;0;EEP 3-A;1/2;French;
audioservice;RADIO PAIS      ;9A;F6C0;11;726;66;88;DAB+;0;EEP 3-A;1/2;French;
audioservice;FUEGO CAP SAO   ;9A;F9F1;5;330;66;88;DAB+;0;EEP 3-A;1/2;French;
Allocating 15 zero-copy buffers
checking data in channel 9B

db TII:
~ $ ls -l /home/user/.txdata.tii 
-rw-r--r-- 1 user user 730416  8 avril 13:54 /home/user/.txdata.tii

I didn't quite understand the prin part :( my development skills are very limited. :/

thanks again !

jeremyp3, 2025-06-27:
It actually works for TII, thanks!

if it's not too time-consuming for you, could you also correct the -d and -D when you have time, so that it works like in the other examples?

after that, I won't bother you anymore! :)

thanks again for your time

JvanKatwijk, 2025-06-27:
That should work, I changed it together with setting the -I flag

This is the essential code
           int  the_timeSyncTime        = timeSyncTime;
           int  the_freqSyncTime        = freqSyncTime;
           int  the_tiiSyncTime         = tiiSyncTime;

JvanKatwijk, 2025-06-27:
> I can confirm that it does work for -I, but I don't reproduce it for -d.
> For example, if I set -d 60 and position myself on an empty dab channel, it
> won't wait the full 60 seconds, it will move on to the next channel after 5
> seconds or so.

If you specify -d 20 and -D 20 you can see - for those channels where
timesync worked but no ensemble was found - that the counter starts at 20
and
steps down one per second

If that is not the case, then you should look at the code and see whether
or not you can find the lines
  while (true) {
           bool firstTime       = true;
           bool firstService    = true;
           int  the_timeSyncTime        = timeSyncTime;
           int  the_freqSyncTime        = freqSyncTime;
           int  the_tiiSyncTime         = tiiSyncTime;

that et the counters

jeremyp3, 2025-06-27:
:~ $ time /tmp/dab-scanner-rtlsdr -C 10A -d 20 -D 20
dab_scanner V 2.0alfa,
[...]
going for rtlsdr 209936000 50
OK, functions seem to be loaded
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
samplerate set to 2048000
[...]
effective gain: gain 25.4
Allocating 15 zero-copy buffers
checking data in channel 10A
Allocating 15 zero-copy buffers
checking data in channel 10B
^C

real	0m7,043s
user	0m2,789s
sys	0m0,264s

jeremyp3, 2025-06-28:
after a test last night with propagation, i can confirm that the -D (attempt to decode a set) works well. there just remains the -d when there is no signal at all on the channel which does not respect the value as shown in my comment above :)

I felt it was important to make this clarification. :)

thanks !!

closed 2025-06-29 by JvanKatwijk

JvanKatwijk, 2025-06-30:
> thanks for the last fix !
> I can confirm that everything works fine now.

To have a -d parameter is actually a stupid idea, the software itself has a
mechanism to see whether or not a DAB signal is
available and viable
(if during one or two seonds no timesync can be found there is no DAB
signal, if during say 5 to 10 tries a  timesync might have been detected
but the
software cannoit identify the proper start address of the DAB frame it is
definitely no DAB signal)

jeremyp3, 2025-06-30:
I understand your point of view, but I don't share it entirely.

when i use eti-cmdline or the examples provided in this repo, i admit to setting the -d to a value between 10 and 20 to make sure there's really no signal at all. it's really useful in my way of doing DXRNT. I agree with you, cases where the signal takes a long time to be seen are rare, but it can happen, especially for distant signals. Once the signal is seen, there's still the decoding part, which works this time with -D, and here too, for distant signals, it can be interesting to set a high value.

Translated with DeepL.com (free version)