JvanKatwijk/dab-cmdline #109: Build Failure - Compatibility with CMake < 3.5 has been removed from CMake

D3vil0p3r, 2025-09-18:
Hey @JvanKatwijk  I'm trying to build dab-cmdline on new distros that use CMake version 4.1 but it seems that the tool still needs old CMake version.

Could you please improve it by making it compatible with the latest version of CMake?

During the build, currently, the error is:
```
==> Starting build()...
CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!
==> ERROR: A failure occurred in build().
```
If I force the usage of  `-DCMAKE_POLICY_VERSION_MINIMUM=3.5`, I get:
```
-- The C compiler identification is GNU 15.2.1
-- The CXX compiler identification is GNU 15.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "2.5.1")
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f, version 3.3.10
-- Found FFTW3F: /lib/libfftw3f.so
-- Found zlib: 
-- Configuring done (0.6s)
CMake Error at CMakeLists.txt:227 (add_library):
  Cannot find source file:

    ./src/backend/charsets.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm
  .ccm .cxxm .c++m .h .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90
  .f95 .f03 .hip .ispc

CMake Error at CMakeLists.txt:227 (add_library):
  No SOURCES given to target: dab_lib

CMake Generate step failed.  Build files cannot be regenerated correctly.
```

JvanKatwijk, 2025-09-19:
charsets was - long time ago - moved to the directory support.
I do not have cmake V4 , so I can change (and I did) the headers in the
CMakeLists.txt files but I cannot test it

D3vil0p3r, 2025-09-19:
@JvanKatwijk I tested it by running:
```
  cmake -B build -S "library" \
  	-Wno-dev \
  	-DCMAKE_BUILD_TYPE='None' \
  	-DCMAKE_INSTALL_PREFIX=/usr
  make -C build
```
but I get this missing file error:
```
-- The C compiler identification is GNU 15.2.1
-- The CXX compiler identification is GNU 15.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "2.5.1")
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f, version 3.3.10
-- Found FFTW3F: /lib/libfftw3f.so
-- Found zlib: 
-- Configuring done (0.7s)
-- Generating done (0.0s)
-- Build files have been written to: /tmp/makepkg/libdab/src/dab-cmdline/build
make: Entering directory '/tmp/makepkg/libdab/src/dab-cmdline/build'
[  5%] Building CXX object CMakeFiles/dab_lib.dir/src/dab-processor.cpp.o
[  5%] Building CXX object CMakeFiles/dab_lib.dir/dab-api.cpp.o
[ 10%] Building CXX object CMakeFiles/dab_lib.dir/src/ofdm/phasereference.cpp.o
[ 10%] Building CXX object CMakeFiles/dab_lib.dir/src/ofdm/ofdm-decoder.cpp.o
[ 12%] Building CXX object CMakeFiles/dab_lib.dir/src/ofdm/phasetable.cpp.o
[ 15%] Building CXX object CMakeFiles/dab_lib.dir/src/ofdm/freq-interleaver.cpp.o
[ 17%] Building CXX object CMakeFiles/dab_lib.dir/src/ofdm/sample-reader.cpp.o
[ 20%] Building CXX object CMakeFiles/dab_lib.dir/src/ofdm/timesyncer.cpp.o
[ 22%] Building CXX object CMakeFiles/dab_lib.dir/src/ofdm/fic-handler.cpp.o
[ 25%] Building CXX object CMakeFiles/dab_lib.dir/src/ofdm/fib-processor.cpp.o
/tmp/makepkg/libdab/src/dab-cmdline/library/src/ofdm/fib-processor.cpp:25:17: fatal error: fib-processor.h: No such file or directory
   25 | #include        "fib-processor.h"
      |                 ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/dab_lib.dir/build.make:205: CMakeFiles/dab_lib.dir/src/ofdm/fib-processor.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:88: CMakeFiles/dab_lib.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
make: Leaving directory '/tmp/makepkg/libdab/src/dab-cmdline/build'
```

closed 2025-10-15 by JvanKatwijk